whatwg/bluetooth #244: How to add disconnect event handler

aircable, 2016-06-14:
I would like to get a disconnect event from the gatt server.

            navigator.bluetooth.requestDevice(options)
                .then(function ( device ) {
                    console.log('Name: ' + device.name);
                    server.addEventListener( 'gattserverdisconnected', handleDisconnect );
                    return device.gatt.connect();
                })
                .then(function ( server ) {
                    console.log('server ' + server );
                    return server.getPrimaryService( MESH_SERVICE_UUID );
                })

The event listener add is accepted, but never executed. The BluetoothRemoteGATTServer object does not have a addEventListener function.
Am I doing something wrong? This is on Android Chrome 51.0.2704.81
Thanks

beaufortfrancois, 2016-06-14:
As shown in https://googlechrome.github.io/samples/web-bluetooth/device-disconnect.html, the `gattserverdisconnected` event must be registered on the BluetoothDevice, not the server.

beaufortfrancois, 2016-06-14:
In your code, here's how it should look like:

```js
        navigator.bluetooth.requestDevice(options)
            .then(function ( device ) {
                console.log('Name: ' + device.name);
                device.addEventListener( 'gattserverdisconnected', handleDisconnect );
                return device.gatt.connect();
            })
            .then(function ( server ) {
                console.log('server ' + server );
                return server.getPrimaryService( MESH_SERVICE_UUID );
            })

aircable, 2016-06-14:
yes, sorry, that was a typo. I tried that and forgot to change it back.
this threw an error of course.

device.addEventListener is not being called. That was my error report.

g-ortuno, 2016-06-14:
As you can see from the [Implementation Status](https://github.com/WebBluetoothCG/web-bluetooth/blob/gh-pages/implementation-status.md), the disconnection event was added in m52. I recommend you download [Chrome Dev](https://play.google.com/store/apps/details?id=com.chrome.dev&hl=en) to try it out.

aircable, 2016-06-14:
that explains it, thanks, will report

aircable, 2016-06-14:
closed, it does work, on Chrome DEV
Thanks

closed 2016-06-14 by aircable