whatwg/bluetooth #245: Playbulb candle app stuck at server.getPrimaryService(CANDLE_SERVICE_UUID)

krantik, 2016-06-27:
I am trying to use the latest Playbulb candle app on Mac/Chrome/Version 51.0.2704.103 (64-bit) After selecting the device the app is stuck. Debugging thru the code, show the call server.getPrimaryService(CANDLE_SERVICE_UUID) never returns. Anything I am missing?

g-ortuno, 2016-06-28:
We currently don't support services and characteristics on macOS m51. We are hoping to get them working on m53. For the latest implementation status you can check [Implementation Status](https://github.com/WebBluetoothCG/web-bluetooth/blob/gh-pages/implementation-status.md).

fyi: issue about the implementation belong in the [chromium issue tracker](https://bugs.chromium.org/p/chromium/issues/list), rather than on the Spec repo.

krantik, 2016-06-28:
Got it thank you.

closed 2016-06-28 by krantik

beaufortfrancois, 2016-06-28:
@krantik And here we are ;) https://plus.google.com/+FrancoisBeaufort/posts/hgYpdoTfXUs