whatwg/bluetooth #251: Problem in getting device info for custom UUIDs in Mac

slashmili, 2016-07-20:
I have an problem with testing `device-info` sample in Mac

I'm testing on macOs 10.12 Beta with Chrome Version 54.0.2800.0 canary (64-bit) 

When I try to get device info form own UUID(713d0100-503e-4c75-ba94-3148f18d941e) which is advertised from a watch(android), I can't find my watch in the pair popup.
(I tested with predefined Bluetooth service and BLE Peripheral Simulator app and I get the device info in Mac)

However when I test the same thing on chromebook, Google Chrome OS Version 52.0.2743.57 beta (64-bit), I get my watch on the popup list and the `device-info` sample works perfectly.

In both cases I already paired the watch with the OS.

Do you have clue what is going wrong in Mac? I can provide more debugging info if it's needed.

beaufortfrancois, 2016-07-20:
Thanks for reporting the issue @slashmili.
May I ask you to [file a Web Bluetooth bug](https://bugs.chromium.org/p/chromium/issues/entry?components=Blink%3EBluetooth) and [add Bluetooth logs](https://www.chromium.org/developers/how-tos/file-web-bluetooth-bugs)?

g-ortuno, 2016-07-20:
As @beaufortfrancois said, thanks for reporting! But this is not really the place to talk about issues with the implementation. This repo is about the spec after all. So in the future please open an issue on crbug.com/new

That said, my guess is that the watch is using Bluetooth Classic which is not supported.

closed 2016-07-20 by jyasskin

slashmili, 2016-07-20:
Sure! I'll to open an issue in chromium tracker. 

And @g-ortuno, no the watch is not using Bluetooth Classic.

beaufortfrancois, 2016-07-21:
Please do so @slashmili. If that's too sensitive, please feel free to drop me an email at `fbeaufort [ a t ] google.com`