whatwg/bluetooth #253: Add a "connectable" field to BluetoothDevice

jracle, 2016-07-26:
As @beaufortfrancois suggests (following @Vudentz remark) in pull-request #239 , it would be valuable to add a `boolean connectable` member to `BluetoothDevice`.

Instances of `BluetoothDevice` return by a _regular_ `requestLEScan` (the one without explicitly filtering out connectable devices) have no hint whether or not we can connect to their GATT server.

Once we get device from `requestDevice`, such a `BluetoothDevice` instance would also hold same `boolean connectable` member. Subsequent call to `.gatt.connect()` would fail if `connectable` is `false`.

jyasskin, 2016-07-27:
What should BluetoothDevice.connectable return on Android, where we don't know if a device is sending connectable advertisements? I think we could leave it `false` or allow it to be `null`.

Also, what do you mean by "be opposed the fact that we can pre-filter devices"?

jracle, 2016-07-28:
Thanks @jyasskin good point! Sorry for missing Android issue.. indeed I'd be happy with a nullable `boolean? connectable`.

Sorry for the "be opposed the fact that we can pre-filter devices" : I meant that we can also say that provided we have a filter to return only connectable devices, then we don't need the boolean.

g-ortuno, 2016-07-28:
I think it might be confusing for developers to have this member if android doesn't support it. Imagine a developer on macOS that writes the following line:
```js
if (device.connectable) {
  return device.gatt.connect();
}
```

That will work while he's developing but fail on real android devices.

jracle, 2016-07-28:
oops yes indeed @g-ortuno ... that doesn't solve our limitations

pzboyz, 2016-07-28:
In each BLE ADV_IND advert there is the Flags AD type (Although it is arguably optional to include it) the Flags AD type also indicates if the device is in pairing mode (because one or both of the discoverable bits are set), it is probably useful if that information is also propagated up to the browser API.

g-ortuno, 2016-07-28:
@pzboyz Do you have any use cases in mind? It would help if we had use cases to know if such a feature is useful or not.

mkv123, 2018-08-20:
The Android Bluetooth API's scan results do indicate if the advertisement indicated connectable or not: https://developer.android.com/reference/android/bluetooth/le/ScanResult.html#isConnectable() . This API is available since Android 8.0 aka API Level 26. 

However, please be aware the the Bluetooth spec allows for a device to interleave connectable and non-connectable advertisement, and in fact this is commonly used with beacons. It may make more sense to stick closer to the spec, and indicate connectability per advertisement packet, rather than per device.

g-ortuno, 2018-08-20:
Oh nice. Agree that having it as part of the adv packet event would be better. We still need to figure out what to return on older versions of Android though.