whatwg/bluetooth #255: Clarify the role of requestDevice vs requestLEScan

jracle, 2016-07-26:
Sorry for eventually re-opening closed discussion here, but I see ongoing discussions around connected peripherals ( #237 #243  #253 ) and scanning ( #191 ), with a pull-request ( #239 )for the latter, planned for [3rd milestone : BLE Scanning](https://github.com/WebBluetoothCG/web-bluetooth/milestones).

Forgetting a bit about this spec (that I'm re-discovering anyway... my concern is to clarify retrieving connected peripherals vs the ones which are just advertizing their services.

In Apple's CoreBluetooth API, [CBCentralManager](https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBCentralManager_Class/index.html) offers clean separation of [retrieving connected peripherals](https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBCentralManager_Class/index.html#//apple_ref/occ/instm/CBCentralManager/retrieveConnectedPeripheralsWithServices:) vs [scanning for peripherals](https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBCentralManager_Class/index.html#//apple_ref/occ/instm/CBCentralManager/scanForPeripheralsWithServices:options:).

Why not limiting `requestDevice` to retrieving connected peripherals only? We would still use `requestLEScan` to discover peripherals we're interested in, then use `requestDevice` to connect to their GATT server (though maybe those steps can be shortened some way).

I'm probably missing something here guys, sorry in advance!

jyasskin, 2016-07-27:
My goal is to have folks who want to connect to a particular device, use `requestDevice()`, and folks who want a stream of advertisements from a bunch of nearby devices, use `requestLEScan()`. The devices returned by `requestLEScan()` should have a way to upgrade to GATT communication, but doing it that way will involve two permission requests (1 for scanning and 1 for the upgrade), while `requestDevice()` will only involve one request.

I don't want to limit `requestDevice()` to already-connected peripherals because that'll force users to find their OS-specific way to connect the device before pairing it to the website. We should let them do that in one step.

I'm happy to add text to the spec clarifying this; let me know what sort of text would help you the most.

jracle, 2016-07-28:
@jyasskin thanks for the clarification, it makes sense to me no pb. Point is that as of today many developers (not web developers I agree) are used to APIs that clearly discriminate retrieving connected peripherals ([Android](https://developer.android.com/reference/android/bluetooth/BluetoothManager.html#getConnectedDevices(int)), [iOS](https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBCentralManager_Class/index.html#//apple_ref/occ/instm/CBCentralManager/retrieveConnectedPeripheralsWithServices: ), which are closer to underlying implementation, and we are trying here to mix 2 concepts into 1 : devices connected to the system versus nearby devices.

At a UX standpoint, suppose I want to retrieve my connected BLE HID devices (e.g. my MX Master mouse), and don't want to be proposed the one from my neighbor, how sould I know which device to select?

As regards discovery, couldn't then we introduce requestLEScan() [from day 1](https://github.com/WebBluetoothCG/web-bluetooth/milestones), so that we'd use it to discover nearby devices. I admit that if we don't have it at the beginning, then requestDevice() also has to perform discovery. My fear is that we might not want to change its behavior once we ship first milestones.

Thanks !

jyasskin, 2016-08-05:
Yeah, `requestDevice()` is a new kind of interface that lots of people won't be familiar with. We're doing it this way so that we can present users a choice of devices to grant access to, instead of making it a yes/no decision for just the device the site wants. We think that'll have better security properties, although we don't have good evidence for that yet.

We do want to add things to the chooser to make it easy to distinguish multiple nearby mice. We're currently adding a note in the Chrome chooser to identify devices that have already been paired, but I think we should add another one for devices that are physically connected.

jracle, 2016-08-05:
> Yeah, requestDevice() is a new kind of interface that lots of people won't be familiar with. 

That's my fear :).

> We're doing it this way so that we can present users a choice of devices to grant access to, instead of making it a yes/no decision for just the device the site wants. We think that'll have better security properties, although we don't have good evidence for that yet.

What prevents us to show a (different) chooser twice (when requesting connected peripherals and when scanning)?

> ..., but I think we should add another one for devices that are physically connected.

Yes, that's indeed needed when choosing my mouse and not my neighbors'.

jyasskin, 2016-08-05:
> What prevents us to show a (different) chooser twice (when requesting connected peripherals and when scanning)?

I don't see a way to make either of those *choosers*: what is there for the user to choose about "scan for peripherals" or "communicate with this particular scanned device"?

stroiman, 2020-06-09:
@jyasskin writes

> The devices returned by requestLEScan() should have a way to upgrade to GATT communication

Is there any guidance as to how to do this?

As far as I can understand from the documentation, I should call `navigator.premissions.request({ name: 'bluetooth', deviceId: event.device.id })`, but that generates an error, "The provided value 'bluetooth' is not a valid enum value of type PermissionName"

odejesush, 2020-06-09:
> @jyasskin writes
> 
> > The devices returned by requestLEScan() should have a way to upgrade to GATT communication
> 
> Is there any guidance as to how to do this?

We haven't figured out how exactly to do this yet, so it's not possible at the moment.

> As far as I can understand from the documentation, I should call `navigator.premissions.request({ name: 'bluetooth', deviceId: event.device.id })`, but that generates an error, "The provided value 'bluetooth' is not a valid enum value of type PermissionName".

The Bluetooth Permissions API integration is also not implemented. The only way to request a device to connect to is to use `navigator.bluetooth.requestDevice()` and provide the services that you would like to connect to in the `filters` field or in the `optionalServices` field.

stroiman, 2020-06-10:
Thanks for the reply 👍. Is this actively being worked on? As this issue is close to 4 years old, I assume an update is not just around the corner.