whatwg/bluetooth #257: Consider devicechange event for when a new granted bluetooth device is made available,

beaufortfrancois, 2016-08-01:
In the same way, `devicechange` event is triggered when a new media input or output device is made available,  it would be nice to add a similar event for when a new granted bluetooth device is made available.

Source: https://groups.google.com/a/chromium.org/forum/#!topic/blink-dev/wdSV0zUuvJQ

```js
navigator.bluetooth.addEventListener('devicechange', event => {
    console.log(event.target.device.id);
    console.log(event.target.device.name);
    console.log(event.target.device.uuids;
    console.log(event.target.device.gatt.connected);
});
```

jracle, 2016-08-03:
@beaufortfrancois what is your use case, please clarify? I guess it is for other UA to be notified ?

When you say 

> hen a new granted bluetooth device is made available

do you mean device is discovered and granted though `requestDevice` ?

Thanks.

beaufortfrancois, 2016-08-04:
> do you mean device is discovered and granted though requestDevice ?

Yes. Instead of using [BroadcastChannel](https://www.chromestatus.com/features/4585496197988352), it would be nice to have this kind of info baked in the API to allow several tabs to sync their states.

I guess the `deviceremoved` event would also be nice.

jracle, 2016-08-04:
@jyasskin thanks for the clarification.

jyasskin, 2016-08-04:
[`navigator.permissions.query()`](https://w3c.github.io/permissions/#dom-permissions-query) returns an object that has [an event that fires when permission has changed](https://w3c.github.io/permissions/#dom-permissionstatus-onchange). I think developers can use that to solve this use case?

beaufortfrancois, 2016-08-05:
For the sake of completeness, is this how it would work?

```js
navigator.permissions.query({
  name: "bluetooth",
  deviceId: sessionStorage.lastDevice /* because we've stored it previously */
}).then(result => {
  result.addEventListener('change', function(event) {
    // Permission status has changed for this device
    console.log(event.target);
  })
})
```

If `result.devices` were to return multiple devices, would there be multiple events fired for each device?

Is this how do I access a newly paired `BluetoothDevice` object with this `change`  triggered event?
```js
navigator.permissions.query({
  name: "bluetooth",
  optionalServices: [ 'battery_service' ], 
}).then(result => {
  result.addEventListener('change', function(event) {
    console.log(event.target); /* Is this BluetoothDevice?
  })
})
```

@jyasskin Sorry for this, I'm still having hard time getting the full `Permissions` API scope for Web Bluetooth.

jyasskin, 2016-08-05:
Close. To detect that you've lost access to the device, use:

```js
navigator.permissions.query({
  name: "bluetooth",
  deviceId: sessionStorage.lastDevice /* because we've stored it previously */
}).then(result => {
  if (result.devices.length == 0) {
    console.log("Already lost access");
  } else {
    assert(result.devices[0].id == sessionStorage.lastDevice);
    result.addEventListener('change', function(event) {
      assert(result == event.target);
      if (result.devices.length == 0)
        console.log("Lost access");
    });
  }
})
```

To learn about newly-paired devices, use:

```js
navigator.permissions.query({
  name: "bluetooth",
  services: [ 'battery_service' ], 
}).then(result => {
  result.addEventListener('change', function(event) {
    for (let device in result.devices) {
      if (!(device.id in knownDeviceIds)) {
        console.log("New device!", device);
      }
    }
  });
})
```

> If `result.devices` were to return multiple devices, would there be multiple events fired for each device?

There would be an event for each time `result.devices` changes, which might be just once if the user revokes all devices in one UI action, or multiple if the user revokes them one at a time.

beaufortfrancois, 2016-08-09:
Based on @jyasskin code samples, I believe `devicechange` events are not necessary. All we have to do is to help developers with official code samples to understand how it works. 
Closing this issue then.
Feel free to reopen if something's not clear.

closed 2016-08-09 by beaufortfrancois