whatwg/bluetooth #258: What is a secure Device Firmware Update Service?

beaufortfrancois, 2016-08-01:
Raised by https://github.com/WebBluetoothCG/registries/issues/10, we need to define what makes a secure Device Firmware Update (DFU) GATT service that will not blacklisted by Web Bluetooth.

This thread is an ongoing discussion whose goal is to list Secure DFU requirements.

gfwilliams, 2016-08-01:
Just IMO, I feel like it's not such a big deal in cases where the user has to do something physical with the device to initiate flashing (eg. holding down a button while inserting the battery). Obviously if flashing can be initiated via Web Bluetooth it's a major security issue (but then it is on all platforms).

As someone doing an Open Source (hardware and software) product I'm stuck in a difficult position... To implement firmware update in a way that won't be blocked I've actually got to lock my users out of their own devices... They'll no longer be able to compile their own firmware, as I obviously can't share the private key.

Do you see any way around this?

gfwilliams, 2016-08-01:
One solution would be to do like browsers currently do with insecure HTTPS - a big `This could damage your device` sort of thing that requires a very deliberate opt in. In feels more in the spirit of an open web than just blocking things outright.

jracle, 2016-08-03:
Hi guys,

  in a nutshell I'm in favor of black-listing potentially dangerous **_standard_** [services](https://www.bluetooth.com/specifications/gatt/services) and [characteristics](https://www.bluetooth.com/specifications/gatt/characteristics), and agree with standard UUIDs that have been chosen so far in [gatt_blacklist.txt](https://github.com/WebBluetoothCG/registries/blob/master/gatt_blacklist.txt). 

  However **_I want to sit against blacklisting vendor-defined services and characteristics_**, and leave responsibility to the vendor for its device being eventually corrupted. As it is already stated in spec, [**_device access is powerful_**](https://webbluetoothcg.github.io/web-bluetooth/#device-access-is-powerful), and user must be aware of it.

  My first caution with the practice of black-listing is that we'd always be behind both standard, and of course vendor-defined UUIDs. As I state above, I'm in favor of such blacklist for standard UUIDs cause standard evolves slowly, and my guess is that dangerous UUIDs, that needs black-listing, have already been put into it. For instance, I don't want any other UI to turn of privacy flag, or to trace my gizmo's serial number..

  Now, as regards vendor-defined services and characteristics, please also pay attention that a service may have multiple purposes, and hence may not be dedicated to, say, solely Device Firmware Update (DFU). 

  At Logitech, we are used to encapsulating our own protocols in standard-defined ones. We use for years a protocol called HID++, which originated from HID vendor. There are 2 versions, [which are public](https://drive.google.com/folderview?id=0BxbRzx7vEV7eWmgwazJ3NUFfQ28&usp=sharing) : version 1.0 and version 2.0. Version 2.0 is _object-oriented_ in spirit, in that it is a set of interfaces (that we call features in our lingo) defining functions and events. Similar to what are services and characteristics in GATT.

  We have defined a vendor service for HID++ over GATT, that will expose various features, among which you'll find DFU. Now the fact that DFU is secure or not lies in its implementation. We guarantee our DFU implementation to be secure, by having implemented the following : 
- DFU file is digitally signed using private key
- Device firmware embeds public key for signature verification during DFU, and hence rejects any requests for injecting rogue firwmare.

So, **please do not black-list vendor-defined services** at risk to block all device access, and make us fall-back to using native apps. Native apps aiming to break device firmware or just play with it are already done by couple of reverse-engineers (we love them and want to make them happy!), and other security-research companies help us to pinpoint and correct defects (e.g. MouseJack / KeyJack for Unifying receiver by [Bastille](https://www.bastille.net/)).

To finish, we might want to black-list DFU services provided in demo kits, but this makes testing being a burden. I'd leave responsibility of device vendors not to use those services, but rather provide their own GATT service.

krantik, 2016-08-03:
I think it shouldn't be the responsibility of web bluetooth to define this black list, it should be part of Bluetooth spec if it is deemed necessary. There are service uuid's defined as part of the Bluetooth spec and there is a conversation to even create a service for firmware update, do we then go in and add that to the black list? There are always going to be cases where some see the service uuid as required, for example someone might choose a service uuid without knowing it was black listed here.

thegecko, 2016-08-03:
Blacklisting known services/characteristics could lead to implementors bypassing the blacklisted IDs by simply using vendor-defined services/characteristics.
This potentially leads to fragmenting the well-known service lists, making it less manageable and maybe less secure.

I agree with @gfwilliams, a warning dialog or similar (with relevant `here be dragons` and `implementor takes no responsibility` caveats) seems to be a better approach which warns the user and doesn't get in the way of implementing awesome stuff with Web Bluetooth.

Could a model similar to the android app store be implemented?

e.g.

`This web page would like to access the following services on your device:`

* Heart Rate
* DFU (Warning, this could break you)

Cancel | Allow |
--- | ---

jyasskin, 2016-08-03:
I think a secure firmware update service has to either require a signature from the old firmware's author or require the user to opt in on the device, like in @gfwilliams' example of holding down a button while inserting the battery.

The difficult decisions come when some implementations of the same service are secure and others aren't. I think the right rule there is that if the initial definition of a service encouraged a secure implementation, then we allow the service, and if the initial definition did not encourage a secure implementation, then we blacklist it. The "definition" doesn't need to be formal: if a service is initially used on a single device in a secure way, and then copied by other devices in an insecure way, I'd lean toward leaving it allowed, and calling the insecure copies a mistake that those devices' creators are responsible for.

I also don't think we should proactively look for insecure services that are only used on a single device. We should wait for those devices' creators to complain to us about a security problem they're encountering in the wild, and ask to have their service blacklisted. The risk here is that they'll have re-used someone else's service without telling us, and we'll break that other device by surprise. Anyone have preferences about how we should avoid that?

This does mean that individual devices can avoid the blacklist by just changing a standard service's UUID. I think that's fine: if a UUID is only used on a single device, it's 1) hard for an attacker to write a single program that attacks a huge number of people's devices, and 2) it's easy for us to blacklist that UUID if the device maker later realizes it was a mistake, since that won't cause collateral damage.

----

A registry of human-readable names for services is hard because there are so many services, but it's plausible for the few services on the blacklist. For the Chrome implementation, we'll have to figure out how to prioritize that. For instance, we probably don't want to delay shipping the API in order to make the blacklist overridable.

nastj333sikan, 2016-08-03:
Получить Outlook для Android

От: Jeffrey Yasskin

Отправлено: Четверг, 4 августа, 1:04

Тема: Re: [WebBluetoothCG/web-bluetooth] What is a secure Device Firmware Update Service? (#258)

Кому: WebBluetoothCG/web-bluetooth

I think a secure firmware update service has to either require a signature from the old firmware's author or require the user to opt in on the device, like in @gfwilliams' example of holding down a button while inserting the battery.

The difficult decisions come when some implementations of the same service are secure and others aren't. I think the right rule there is that if the initial definition of a service encouraged a secure implementation, then we allow the service, and if the initial definition did not encourage a secure implementation, then we blacklist it. The "definition" doesn't need to be formal: if a service is initially used on a single device in a secure way, and then copied by other devices in an insecure way, I'd lean toward leaving it allowed, and calling the insecure copies a mistake that those devices' creators are responsible for.

I also don't think we should proactively look for insecure services that are only used on a single device. We should wait for those devices' creators to complain to us about a security problem they're encountering in the wild, and ask to have their service blacklisted. The risk here is that they'll have re-used someone else's service without telling us, and we'll break that other device by surprise. Anyone have preferences about how we should avoid that?

This does mean that individual devices can avoid the blacklist by just changing a standard service's UUID. I think that's fine: if a UUID is only used on a single device, it's 1) hard for an attacker to write a single program that attacks a huge number of people's devices, and 2) it's easy for us to blacklist that UUID if the device maker later realizes it was a mistake, since that won't cause collateral damage.

A registry of human-readable names for services is hard because there are so many services, but it's plausible for the few services on the blacklist. For the Chrome implementation, we'll have to figure out how to prioritize that. For instance, we probably don't want to delay shipping the API in order to make the blacklist overridable. 

—

OpenSSL Security Advisory [3rd May 2016]
========================================

Memory corruption in the ASN.1 encoder (CVE-2016-2108)
======================================================

Severity: High

This issue affected versions of OpenSSL prior to April 2015. The bug
causing the vulnerability was fixed on April 18th 2015, and released
as part of the June 11th 2015 security releases. The security impact
of the bug was not known at the time.

In previous versions of OpenSSL, ASN.1 encoding the value zero
represented as a negative integer can cause a buffer underflow
with an out-of-bounds write in i2c_ASN1_INTEGER. The ASN.1 parser does
not normally create "negative zeroes" when parsing ASN.1 input, and
therefore, an attacker cannot trigger this bug.

However, a second, independent bug revealed that the ASN.1 parser
(specifically, d2i_ASN1_TYPE) can misinterpret a large universal tag
as a negative zero value. Large universal tags are not present in any
common ASN.1 structures (such as X509) but are accepted as part of ANY
structures.

Therefore, if an application deserializes untrusted ASN.1 structures
containing an ANY field, and later reserializes them, an attacker may
be able to trigger an out-of-bounds write. This has been shown to
cause memory corruption that is potentially exploitable with some
malloc implementations.

Applications that parse and re-encode X509 certificates are known to
be vulnerable. Applications that verify RSA signatures on X509
certificates may also be vulnerable; however, only certificates with
valid signatures trigger ASN.1 re-encoding and hence the
bug. Specifically, since OpenSSL's default TLS X509 chain verification
code verifies the certificate chain from root to leaf, TLS handshakes
could only be targeted with valid certificates issued by trusted
Certification Authorities.

OpenSSL 1.0.2 users should upgrade to 1.0.2c
OpenSSL 1.0.1 users should upgrade to 1.0.1o

This vulnerability is a combination of two bugs, neither of which
individually has security impact. The first bug (mishandling of
negative zero integers) was reported to OpenSSL by Huzaifa Sidhpurwala
(Red Hat) and independently by Hanno Böck in April 2015. The second
issue (mishandling of large universal tags) was found using libFuzzer,
[... 127 lines cut ...]
fix was developed by Matt Caswell of the OpenSSL development team.

Note
====

As per our previous announcements and our Release Strategy
(https://www.openssl.org/policies/releasestrat.html), support for OpenSSL
version 1.0.1 will cease on 31st December 2016. No security updates for that
version will be provided after that date. Users of 1.0.1 are advised to
upgrade.

Support for versions 0.9.8 and 1.0.0 ended on 31st December 2015. Those
versions are no longer receiving security updates.

References
==========

URL for this Security Advisory:
https://www.openssl.org/news/secadv/20160503.txt

Note: the online version of the advisory may be updated with additional details
over time.

For details of OpenSSL severity classifications please see:
https://www.openssl.org/policies/secpolicy.html

jracle, 2016-08-04:
@jyasskin our devices at Logitech don't always have a physical power switch nor a removable battery or a button. Hence sometimes firmware will transition from application to bootloader (i.e. enter DFU mode) without requiring any action on device.

Also leaving hardware vendors complain about security when they have issues with their services is the best way to go, provided you can give them a way to do it.

jyasskin, 2016-08-04:
@jracle I think Logitech is fine because your devices check that the firmware is signed.

The best way to complain is to file a bug in https://github.com/WebBluetoothCG/registries. It's not the easiest place to discover right now. Where do you think hardware vendors will look when trying to complain?

jracle, 2016-08-05:
@jyasskin thanks!

I don't have a better place than [https://github.com/WebBluetoothCG/registries](https://github.com/WebBluetoothCG/registries) for now. I don't know if a similar registry exists in other specified APIs, any clue?

jyasskin, 2016-08-05:
We're the first that I know of.

jyasskin, 2016-09-17:
Check out https://github.com/WebBluetoothCG/registries/pull/16 where I'm trying to write down the guidelines we want to follow.