whatwg/bluetooth #274: Don't fire characteristicvaluechanged for Read Responses

Emill, 2016-08-16:
Now characteristicvaluechanged is fired both for read responses and notifications. I don't think results of readValue should be sent to the characteristicvaluechanged event. It's better to just return the value in the promise. That makes it easier to write code that uses GATT reads, because you can then easier relate a read response to a particular read request.

Also, now as far as I can see, it's impossible to distinguish a characteristicvaluechanged that happened due to a read from one that happend due to a notification, which might be important when you want to implement something on top of GATT.

The term "characteristic value changed" also does not apply really to the read procedure; the value hasn't changed, you just got a copy of what's stored in the peripheral. For notifications however, the value has usually changed when a notification is sent.

g-ortuno, 2016-08-16:
> Also, now as far as I can see, it's impossible to distinguish a characteristicvaluechanged that happened due to a read from one that happend due to a notification, which might be important when you want to implement something on top of GATT.

I don't think we would be able to implement this given that macOS follows pretty much the same pattern. https://developer.apple.com/reference/corebluetooth/cbperipheraldelegate/1518708-peripheral?language=objc

> The term "characteristic value changed" also does not apply really to the read procedure; the value hasn't changed, you just got a copy of what's stored in the peripheral. For notifications however, the value has usually changed when a notification is sent.

I don't think this was meant to imply that the actual value has changed but rather that we *updated* our copy of the value. Maybe `characteristicvalueupdated` would be more appropriate.

Emill, 2016-08-16:
> I don't think we would be able to implement this given that macOS follows pretty much the same pattern. https://developer.apple.com/reference/corebluetooth/cbperipheraldelegate/1518708-peripheral?language=objc

Yeah it sucks that Apple did it this way. We have reported this to them as well but they really didn't seem to care. It would be bad to be forced to make it worse for the other platforms that Web Bluetooth run on just because Apple has made this very strange decision.

I really think this pattern is very odd. You expect to get the read response as a response to a request, not as some event. Event listeners are usually for something that can be triggered any time. Read responses are not triggered any time, they are only triggered as a response to read requests.

How do you anyway implement, on top of CoreBluetooth, at https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothremotegattcharacteristic-readvalue, the step "Resolve promise with this.value."? If a notification arrives before the read response, do you then incorrectly resolve the promise with the notification value, rather than the read response value that will come in later?

Until Apple decides to fix their API, one could use a workaround that, after a read request is issued, assume the first `didUpdateValueForCharacteristic` that comes in is the response to the read request. Further `didUpdateValueForCharacteristic` are assumed to be notifications. For all other platforms such as Android, Windows, Linux no workarounds would be needed since they have done it the "correct way".

> I don't think this was meant to imply that the actual value has changed but rather that we updated our copy of the value. Maybe `characteristicvalueupdated` would be more appropriate.

If the event is only triggered due to notifications/indications I think both names work. If it handles both notifications/indications and read responses, `characteristicvalueupdated` would be more appropriate.

beaufortfrancois, 2016-08-17:
When characteristic only supports READ GATT operations, it is actually a cool pattern. See https://googlechrome.github.io/samples/web-bluetooth/read-characteristic-value-changed.html for instance.

However I understand why it is confusing... we could either:

- Add a property to the `characteristicvaluechanged`/`characteristicvalueupdated` event:

```js
characteristic.addEventListener('characteristicvalueupdated',
                                 handleCharacteristicValueUpdated);

function handleCharacteristicValueUpdated(event) {
  console.log(event.isReadResponse) /* Returns true if stored value comes from a read operation */
  console.log(event.target.value); /* Characteristic value */
}
```

- Or add a new property to the characteristic object itself which indicates if stored value comes from a read or a notification:

```js
characteristic.addEventListener('characteristicvalueupdated',
                                 handleCharacteristicValueUpdated);

function handleCharacteristicValueUpdated(event) {
  console.log(event.target.isRead) /* Returns true if stored value comes from a read operation */
  console.log(event.target.value); /* Characteristic value */
}
```

I would lean towards first option personally.

Emill, 2016-08-17:
Hehe I see your point if you first want to do a read to get the initial value and then listen for notifications to trigger the same callback when it's updated.

Anyway, your first option looks got to me.

jyasskin, 2016-08-17:
I would also prefer @beaufortfrancois' first option if we make a change.

@Emill, what's your use case for distinguishing reads from notifications in the `characteristicvaluechanged` event? I want to try to avoid changing things based on just "might be important", but it seems like you probably actually ran into this when trying to do something concrete, so I'd like to explore what the concrete thing is. 

Guessing, it seems like to run into this you'd need to be using the same characteristic for two different purposes, perhaps returning unrelated values from a read vs a notification. If I've got that right, why is that the right design instead of using two different characteristics?

Emill, 2016-08-17:
Hi.

If you just read the Bluetooth Core 4.2 specification "4.8.1 Read Characteristic Value", it is a standalone procedure which contains a read request and a read response. It does not seem to make any sense to "merge" the response value into the notification procedure.

Actual use cases may be anything that is using GATT for more advanced things. One might think that people only use GATT for basic stuff like "hard rate monitor" and "battery status" but that is far from the only thing what people do. Actually I'd guess almost all products on the market have at least one GATT service that is being used in a way GATT was not designed for. There are many products that have developed protocols that run on top of GATT to fulfill their needs. Examples are firmware update services (https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v8.x.x/doc/8.1.0/s110/html/a00105.html), serial port emulating services (http://support.dialog-semiconductor.com/resource/da14580-dialog-serial-port-service-source-code), cryptographic protocols (https://github.com/aanon4/HomeKit). A firmware update service is probably the most common one among "advanced" GATT services. For these often the GATT structure does not make much sense, but they are limited to using GATT since that is the only way to communicate with a BLE device. For these protocols all GATT operations does not really mean "read the current attribute value" or "store this value in this characteristic", but rather GATT is only used for transporting or streaming data, so also getting a cached value if the device is disconnected for example does not make any sense. I'm currently aware of at least one cryptographical protocol over BLE that uses reads and notifications on the same characteristic in a totally different way. Once a read request is sent out it expects a read response containing a value with a specific format (encrypted in a specific way). Notifications are also enabled on the same characteristic but they have a different encryption and if it doesn't pass the decryption, the link is dropped.

Having multiple characteristics as you say, while it sometimes makes more sense to have one, only as a workaround for some BLE APIs, would also increase complexity in firmware + lead to longer service discovery times.

So, because I think it is no strange at all to be able to distinguish read responses from notifications, I see no reasons not to do so. I see this as a great opportunity to fix the spec _before_ the final version has become standard :)

jyasskin, 2016-08-17:
I care a lot more about particular products than generic speculation. When one appears, it'll be straightforward to add the event property later.

Emill, 2016-08-17:
Well, at least I would say the opposite - it's easier to do changes to a draft than an implemented standard ;)

Emill, 2016-12-20:
Why was this closed?

Anyway, I could possibly live with the current solution, but then I would like to change the order of
> 3. Fire an event named characteristicvaluechanged with its bubbles attribute initialized to true at this.
> 4. Resolve promise with this.value.

Now it's impossible to know in the characteristicvaluechanged event if the event was sent due to a read response. If the order is reversed, one can be sure that after the promise is resolved, the next characteristicvaluechanged event will correspond to the read response.

beaufortfrancois, 2016-12-26:
I'll let @jyasskin reply on this one.

beaufortfrancois, 2017-01-16:
Maybe @scheib can comment on this one while @jyasskin & @g-ortuno are temporarily unavailable.

g-ortuno, 2017-02-02:
@jyasskin since he was OOO

jyasskin, 2017-02-03:
@Emill, have you come up with a concrete use case where you need to distinguish notifications from read responses? I still think that adding a field to an event is something we can easily do even after everyone's implemented the spec without it.

Re changing the order of "fire an event" and "resolve promise", ["fire an event"](https://dom.spec.whatwg.org/#concept-event-fire) is synchronous, while "resolve promise" is first observable at the next microtask checkpoint, so writing them in the opposite order in the spec would result in the same sequence of user-visible actions (except for further promises resolved in the event handlers). We could queue a task to fire the event, but that risks things getting further out of order, and is different from how most events are fired on the web platform.

Emill, 2017-02-03:
Ok I see. It seems harder than I thought to get the ordering right when you have both promises and events. I think adding the field "read response or notification" to the event would be OK to solve the problem.

But I however still don't really get the logic why the characteristicvaluechanged is fired when a read response is returned and see no really good use case for it either ;)
Even if https://googlechrome.github.io/samples/web-bluetooth/read-characteristic-value-changed.html shows some use of it for polling battery status and similar - that must really be the only use case. But if one really wants to do the same thing when a notification and a read response arrives, one can just setup a function which is called both when a promise resolves as when the event is triggered.

Most times you want to handle the read value in some completion handler for the read request and not handle the general event. So it is strange that the result is delivered in two different ways.
Just because the read operation completes doesn't mean the value has been changed, so the name is not so good either if it's used for read responses. Android has different onCharacteristicRead and onCharacteristicChanged callbacks and nobody complains about that. Windows also delivers read results as a return value to the blocking (or async) read request function and notifications to a dedicated event handler (without read responses), which is exactly how a user would expect it to work and maps very well to the original GATT standard. I have actually seen no other bluetooth stacks that work in the way the current spec is defined, except CoreBluetooth that merges them into same.

Actually our Flic protocol treats read responses and notifications in completely different way and is supposed to be dispatched to different parsers, but since we ignore invalid data it still works with web bluetooth. But I've seen other proprietary protocols as well where reading a value means something completely different than when a notification/indication arrives on the same characteristic. You also have the use case where someone wants to implement yet another gatt wrapper layer on top of web bluetooth - then that platform might define the api differently so you need to know if the value is the result of a read response or a notification/indication.

You should note that even though at first glance one might think everyone uses GATT "as it is supposed to be", i.e. a server with an attribute db a client can read and write to, most larger consumer BLE products you can buy in stores I've seen always use custom characteristics with some binary protocol. The "glucose service", "cycling speed" etc were just something Bluetooth SIG came up with and hoped that they would be used :) but everyone has their own custom profiles just because almost everyone needs custom functionality. Just look at Fitbit's https://pewpewthespells.com/blog/fitbit_re.html, https://devzone.nordicsemi.com/documentation/nrf51/4.4.1/html/group__dfu__ble__service__spec.html or Apple's Apple Watch's protocol. Of course the "heart rate" services and similar are very good for testing out BLE functionality and for example when showcasing Web Bluetooth, but real products are more advanced than that.

I think the web bluetooth standard is great in almost every point and has by far the easiest-to-use API I've ever seen of bluetooth stacks, but this little simple detail just keeps bothering me...

g-ortuno, 2017-02-03:
Though I agree that differentiating between Read responses and Notifications would be nice, I feel we can't have a good enough implementation to guarantee that we will correctly mark the events. More specifically, in macOS there is no difference between Read and Notification. So we basically mark a value changed event as 'read response' if there is a pending read request. You could imagine a scenario where a read request is dispatched and a notification immediately arrives. In this case the event would be mark as a read response even though it is in fact a notification.

Emill, 2017-02-03:
Yes, CoreBluetooth is not so good in this case. But both solutions handles this wrong, i.e. you can get a read request's promise resolved with a notified value rather than the correct read response. We've actually sent a bug report to Apple telling them that it's impossible to distinguish between a read response and a notification.

By the way, noble uses a lower level mechanism in Mac OS X to really solve this problem: https://github.com/sandeepmistry/noble/blob/cbe0213e5f9d02ce40e8873a44f1f9008c0d4397/lib/mac/yosemite.js#L522. It seems there is actually an isNotification parameter that is removed by the outer API.

jyasskin, 2017-02-21:
Sorry for dropping this issue again. Flic's use of different formats over notification vs read counts as a use case in my mind, so that's a good argument to expose the difference. Thanks for pointing out that we can go down to the xpc layer to get at this bit. It's going to take a while for us to implement that on Chromium, so don't hold your breath for this to change, but I now think it's the right long-term direction.

sandeepmistry, 2018-01-31:
>Thanks for pointing out that we can go down to the xpc layer to get at this bit. It's going to take a while for us to implement that on Chromium, so don't hold your breath for this to change, but I now think it's the right long-term direction.

Unfortunately in macOS High Sierra, the `kCBMsgArgIsNotification` XPC parameter is no longer sent now as noted by @jacobrosenthal in https://github.com/sandeepmistry/noble/pull/727. Use of it will be deprecated in the next version of noble.