whatwg/bluetooth #277: writevalue should not update value

scheib, 2016-08-29:
As discussed in https://bugs.chromium.org/p/chromium/issues/detail?id=614534, writing to a characteristic does not neccesarily mean that subsiquent reads will return that value. That behavior is peripheral dependent.

I believe the [writevalue](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothremotegattcharacteristic-writevalue) line should be deleted:

> Set this.value to a new DataView wrapping a new ArrayBuffer containing bytes.

This applies to characteristics and descriptors.

jyasskin, 2016-08-29:
That makes sense. Do we have feedback from outside our team that this is the right thing to do? If not, could you ping public-web-bluetooth and try to get some?

beaufortfrancois, 2016-08-31:
Can you confirm I'll have to update https://googlechrome.github.io/samples/web-bluetooth/link-loss.html to something like this?

```diff

function onWriteButtonClick() {
  if (!alertLevelCharacteristic) {
    return;
  }
  log('Setting Alert Level...');
  let value = document.querySelector('#alertLevelValue').value;
  alertLevelCharacteristic.writeValue(new Uint8Array([value]))
  .then(_ => {
    log('> Alert Level changed to: ' +
-        getAlertLevel(alertLevelCharacteristic.value));
+        getAlertLevel(value));
  })
  .catch(error => {
    log('Argh! ' + error);
  });
}

g-ortuno, 2016-08-31:
Yup. That would work

beaufortfrancois, 2016-09-02:
https://codereview.chromium.org/2287273002 has been pushed.

reillyeon, 2018-09-05:
The spec change and Chromium CL have both been merged.

closed 2018-09-05 by reillyeon