whatwg/bluetooth #284: Add Web Bluetooth API support to change MTU

beaufortfrancois, 2016-09-02:
I've noticed Chrome OS and Android have different maximum transmission unit (MTU) when investigating https://bugs.chromium.org/p/chromium/issues/detail?id=643591 bug and I was wondering if we could add support for requesting an MTU size used for a given connection in the Web Bluetooth API.

Android already has this API which is really useful in my own opinion: https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#requestMtu(int)

WDYT?

beaufortfrancois, 2016-09-02:
@Vudentz Is this something we could do in BlueZ?

jyasskin, 2016-09-02:
I haven't investigated this in the BT spec yet, but why can't we auto-negotiate the largest supported MTU?

beaufortfrancois, 2016-09-02:
In the short term, that would be nice indeed.

Vudentz, 2016-09-05:
BlueZ actually attempts to do the maximum possible, our default per channel is actually 672 bytes given that comes from L2CAP, so in fact it can be over 512 bytes but usually the peripheral limits the MTU. As for exposing an API to set the MTU this is quite tricky because of the following:

1. The MTU shall only be negotiated once, so we might have to fail if that was already done previously.
2. MTU needs to be symmetric for input and output so it might be forced to be lower than the value required.
3. Usually the MTU is used to define application level fragmentation, stream/firmware transfers, etc. These are not very nice to be done over D-Bus and instead perhaps should done over e.g. a pipe passing a fd to the application where one can get the MTU of the connection and perhaps even lock the attribute so it is only accessed by that application using the fd. Note this type of application is abusing the ATT/GATT stack since the attributes were never meant to be used as stream endpoints, instead L2CAP Channels should have been used but it seems most OSes never implemented support for them leading to the current situation.

beaufortfrancois, 2016-09-05:
For the sake of completeness, here are my findings regarding MTU size I've gathered from https://beaufortfrancois.github.io/sandbox/physical-web/fatbeacon/

Chrome OS | Android | Mac
 :---------------: | :-----------: | :------:
512 bytes | 20 bytes | 199 bytes

PR #293 (2016-09-19): Specify that UAs should negotiate the MTU.
Fixes #284 

Preview at https://api.csswg.org/bikeshed/?url=https://raw.githubusercontent.com/jyasskin/web-bluetooth-1/max-mtu/index.bs#dom-bluetoothremotegattserver-connect.

@beaufortfrancois @Vudentz
commit fc36a62b (2016-09-19): Specify that UAs should negotiate the MTU. (#293)

Fixes #284
closed 2016-09-19 by jyasskin