whatwg/bluetooth #290: How to handle changing advertised UUIDs

g-ortuno, 2016-09-08:
Step 8 of [get the BluetoothDevice representing](https://webbluetoothcg.github.io/web-bluetooth/#get-the-bluetoothdevice-representing)  states:
> 8. Add device’s advertised Service UUIDs to result.[[unfilteredUuids]].

But it isn't clear what advertised Service UUIDs means when a device is alternating UUIDs. Is it only the last Advertised UUIDs? All the UUIDs we've seen during that particular scan or all the UUIDs we've seen so far across scans?

Given that only BlueZ caches the advertised UUIDs I think this should be specified.

beaufortfrancois, 2016-09-08:
Background at https://bugs.chromium.org/p/chromium/issues/detail?id=645023

jyasskin, 2016-09-08:
Which do you think is the right set of UUIDs? The field should probably converge on the discovered set of UUIDs that are actually in the GATT server, but for its initial value, maybe it should be the UUIDs the UA has seen "recently", where we leave "recently" up to the UA?

jyasskin, 2016-09-08:
Or should we remove `uuids` entirely, and have people call `getPrimaryServices()` instead?

beaufortfrancois, 2016-09-09:
I actually thought about that yesterday and didn't see the value of keeping `uuids` when we get them with `getPrimaryServices`. I'd vote for removing them.

closed 2016-09-26 by jyasskin

beaufortfrancois, 2016-09-30:
I can't find the chromium bug for removing BluetoothDevice uuids. I think it should be a `ship blocking` issue. WDYT?

scheib, 2016-10-05:
Chromium issue [Remove BluetoothDevice.uuids attribute.](https://bugs.chromium.org/p/chromium/issues/detail?id=653317) filed.

PR #292 (2016-09-26): Remove the BluetoothDevice.uuids field.
It was hard to define the value of this field when a device's
advertisements may list varying sets of services and when the primary
services in the GATT server may be discovered gradually. Developers can
replace their uses of this field with uses of getPrimaryServices().

Fixes #290 

Preview at https://api.csswg.org/bikeshed/?url=https://raw.githubusercontent.com/jyasskin/web-bluetooth-1/remove-uuids/index.bs. Please double-check that there's nothing else I can remove now that this field is gone.
commit 4860b2dd (2016-09-26): Remove the BluetoothDevice.uuids field. (#292)

It was hard to define the value of this field when a device's
advertisements may list varying sets of services and when the primary
services in the GATT server may be discovered gradually. Developers can
replace their uses of this field with uses of getPrimaryServices().

Fixes #290