whatwg/bluetooth #294: Support for requesting devices with specific service data in adv packets

irgu, 2016-09-13:
Is there any plan to add filtering by specific service data? I mean something like regex filter for requesting devices with specific pattern inside the service data.

beaufortfrancois, 2016-09-13:
Current implementation for filtering service data and manufacturer data is pretty simple. However, we've talked about adding masks as you can read at https://github.com/WebBluetoothCG/web-bluetooth/issues/153#issuecomment-134648172

May I ask what is your use case?

irgu, 2016-09-17:
Mentioned comment perfectly applies to what I'm looking for. In my case I have proprietary application level packets encapsulated in service data.

beaufortfrancois, 2016-09-19:
@jyasskin Should we add masks to the spec now then?

beaufortfrancois, 2016-09-23:
@irgu Feel free to comment on https://github.com/WebBluetoothCG/web-bluetooth/pull/297

closed 2016-09-27 by jyasskin

beaufortfrancois, 2021-05-18:
@irgu Is this still something you're interested in?

PR #297 (2016-09-27): Accept filters on manufacturer and service data.
Previews at https://api.csswg.org/bikeshed/?url=https://raw.githubusercontent.com/jyasskin/web-bluetooth-1/masked-data-prefixes/index.bs and https://api.csswg.org/bikeshed/?url=https://raw.githubusercontent.com/jyasskin/web-bluetooth-1/masked-data-prefixes/scanning.bs.

Fixes #294. @irgu
commit 36a5e517 (2016-09-27): Accept filters on manufacturer and service data. (#297)

This should take advantage of WebIDL open dictionaries when they're spec'ed, but we're not waiting on https://www.w3.org/Bugs/Public/show_bug.cgi?id=20158.

Fixes #294