whatwg/bluetooth #296: Clarify undocumented behaviors

dati91, 2016-09-19:
Hi!

There are some cases which not mentioned in the spec:
- [ ] No adapter found
- [ ] Advertised device name is longer than 29 bytes

For these cases Chrome and Servo behaves differently. Chrome throws 'NotFoundError', Servo throws 'TypeError' with an explanation.

This should be unified in the specification.

jyasskin, 2016-09-19:
1. If there's no Bluetooth adapter, then [scan for devices](https://webbluetoothcg.github.io/web-bluetooth/#scan-for-devices) does nothing in steps 3 and 4, and winds up returning an empty list. Then [request Bluetooth devices](https://webbluetoothcg.github.io/web-bluetooth/#requestDevice-prompt) prompts the user to choose one of an empty set of devices, and since there aren't any devices, the user can't pick any, and must cancel, resulting in an empty list. Chrome [short-circuits the prompt](https://cs.chromium.org/chromium/src/content/browser/bluetooth/web_bluetooth_service_impl.cc?l=339), since it already knows the answer. [`requestDevice()`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetooth-requestdevice) turns the empty list of results into a `NotFoundError`.<br>
We just specified an [API](https://webbluetoothcg.github.io/web-bluetooth/#availability) to explicitly check if there's an adapter, so pages can avoid showing UI that doesn't make sense without any Bluetooth support.
2. I assume you mean that a filter for an advertised name demands a name longer than 29 bytes. In that case, [request bluetooth devices](https://webbluetoothcg.github.io/web-bluetooth/#request-bluetooth-devices) says to throw a `TypeError` if it's longer than 248 bytes, but leaves it alone if it's a length that can't currently be advertised. The rest of the sequence is the same as above: you get an empty set of devices since no advertisement can currently include a name longer than 29 bytes, meaning the user has to cancel the chooser, leading to a `NotFoundError`. If Bluetooth 5 increases the advertising limit, longer names could start working without any change to the spec.

So, I think Chrome's behavior is closer to the spec here, although we could definitely change the spec if there's a reason to prefer Servo's behavior.

dati91, 2016-09-20:
Our main concern is that these behaviors are not mentioned in the spec. As you described above the `NotFoundError` seems reasonable.

jyasskin, 2016-09-20:
I got opposite messages from your two sentences. :confused: Do you mean that my explanation of `NotFoundError` convinces you that the spec is clear enough, or that we should add something to the spec to say it more explicitly? If we should add something, can you suggest some text that would help?

dati91, 2016-09-20:
I meant we should change Servo's TypeErrors to NotFoundErrors, but i still think we should add these errors as steps in the spec.

PR #306 (2016-10-10): Allow the UA to skip the chooser if no devices could appear in it.
Preview at https://api.csswg.org/bikeshed/?url=https://raw.githubusercontent.com/jyasskin/web-bluetooth-1/specify-early-request-failures/index.bs#request-bluetooth-devices

Fixes #296 

@dati91
commit 65828536 (2016-10-10): Allow the UA to skip the chooser if no devices could appear in it. (#306)

Fixes #296
closed 2016-10-10 by jyasskin