whatwg/bluetooth #305: Consider allowing UAs to omit navigator.bluetooth on platforms that don't support it.

jyasskin, 2016-10-06:
Some platforms don't support Bluetooth LE at all. Others support it conditionally depending on whether a radio is attached. [`getAvailability()`](https://webbluetoothcg.github.io/web-bluetooth/#availability) covers the second group of platforms, and can be used to cover the first, but it might be simpler for developers to be able to treat no-Bluetooth platforms they same as they treat no-Bluetooth browsers.

https://github.com/WICG/web-share/pull/8 and https://github.com/whatwg/notifications/issues/81 propose this strategy for two other specs.

@slightlyoff @PaulKinlan @RByers

RByers, 2016-10-07:
Agreed that it's more predictable for developers if the API doesn't show up at all in the cases we know the feature is never available on the platform.

beaufortfrancois, 2016-10-10:
Just to be clear, Bluetooth may seem to be unavailable on the platform.
However it could work with cheap Bluetooth USB dongles such as https://www.amazon.com/Medialink-Bluetooth-Adapter-Energy-Technology/dp/B004LNXO28 .

jyasskin, 2016-10-10:
I'm thinking about removing `navigator.bluetooth` on platforms like Windows 7 or MacOX 10.9, where even if the user plugged in a cheap dongle, we still wouldn't be able to give them Web Bluetooth. On platforms that support Bluetooth but machines where they'd have to plug in a dongle, the site would need to use `getAvailability()`.