whatwg/bluetooth #312: Make RSSI non optional

g-ortuno, 2016-10-12:
I think every Advertisement packet comes with an RSSI ([Android](https://developer.android.com/reference/android/bluetooth/le/ScanResult.html), [macOS](https://developer.apple.com/reference/corebluetooth/cbcentralmanagerdelegate/1518937-centralmanager?language=objc)). We should mark the rssi property as always present in [BluetoothAdvertisingEvent](https://webbluetoothcg.github.io/web-bluetooth/#bluetoothadvertisingevent)

jyasskin, 2016-10-13:
Windows? If so, great!

g-ortuno, 2016-10-13:
In 8.1 there are no advertisements because there is no support for scanning. The windows 10 APIs do include RSSI: https://msdn.microsoft.com/en-us/library/windows/apps/windows.devices.bluetooth.advertisement.bluetoothleadvertisementreceivedeventargs.aspx

beaufortfrancois, 2016-10-13:
Shall we include this in our Windows notes? 

```diff
 * [Windows](https://crbug.com/507419): Requires Windows 8.1.
   * To discover devices the user hasn't yet manually paired, requires Windows 10.
+  * The Scanning API will require Windows 10.

mkv123, 2018-08-20:
Btw, if RSSI is not available on some platform, there is a magic value for indicating this specified in the Core Bluetooth spec, RSSI +127. (Admittedly, this is likely intended to be used when the RSSI is temporarily not available.) The magic value can be used both in the case when reading the RSSI of a scan result, as well as the case for reading a connection's RSSI. 

Relevant parts of the spec:

 - Core 5.0, Vol 2. Part E, 7.7.65.2 "LE Advertising Report Event"
 - Core 5.0, Vol 2. Part E, 7.5.4 "Read RSSI Command"