whatwg/bluetooth #313: Error out earlier if no allowedServices.

g-ortuno, 2016-10-12:
A developer was using a namePrefix filter and forgot the optionalServices field, which resulted in getPrimaryServices rejecting with NotFoundError. I'm adding a check for allowedUuids before querying for services in the implementation. Is this something we should include in the getPrimaryService(s) algorithms?

beaufortfrancois, 2016-10-12:
Is it the same as https://chromium.googlesource.com/chromium/src/+/b325b3a86985fc17904c82f712acf41b6a96defb%5E%21/#F10?

shawt, 2016-10-12:
> Is it the same as https://chromium.googlesource.com/chromium/src/+/
> b325b3a86985fc17904c82f712acf41b6a96defb%5E%21/#F10?

Yes. Sorry. I thought they were two different forms.

Trevor

-- 
******************************
Trevor Shaw
Director of Technology
Dwight-Englewood School
315 E. Palisade Ave
Englewood, NJ 07631
"Challenging Minds for a Changing World"
v 201.227.3178
*******************************

g-ortuno, 2016-10-12:
That error shows up only if you call `getPrimaryService(s)` with a UUID. If you call `getPrimaryServices()` and you haven't requested access to any services then you just get `DOMException: No Services found in device.`

beaufortfrancois, 2016-10-14:
Oh you're right! This would be nice to add this. Did you already file a bug for this?

g-ortuno, 2016-10-17:
http://crbug.com/656458