whatwg/bluetooth #324: Advise performing a web privacy assessment

scheib, 2016-10-27:
@lknik suggests in https://github.com/w3ctag/spec-reviews/issues/139#issuecomment-256786859, to add an advisement to the specification that clients of Web Bluetooth perform a web privacy assessment.

jyasskin, 2017-01-20:
@lknik, could you send a PR showing roughly what you'd like to see?

lknik, 2017-01-21:
Hi,

Sure. First, please include a suggestion of "web privacy impact assessment". For example 
"_Web applications that use Web Bluetooth should be subject to a privacy impact assessment of the application taking all aspects of the application into consideration._"?

For any others, i.e. updating Considerations - I'm also happy to help.

lknik, 2018-07-26:
Hi,

Any update around this issue?