whatwg/bluetooth #333: When disconnect is called, [[activeAlgorithms]] is being cleared twice

g-ortuno, 2016-11-03:
Maybe just comment that Cleanup disconnected devices will stop pending connection requests.

jyasskin, 2017-01-20:
Clearing `[[activeAlgorithms]]` in [step 1](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothremotegattserver-disconnect) serves to abort any active `connect()` calls. Step 2 returns early from this algorithm if the device isn't already connected, so we never get to [clean up the disconnected device](https://webbluetoothcg.github.io/web-bluetooth/#clean-up-the-disconnected-device) in that case. It might make sense to have [Clean up ...](https://webbluetoothcg.github.io/web-bluetooth/#clean-up-the-disconnected-device) return early for disconnected devices in order to remove the apparent redundancy. I haven't checked if that has any other unwanted side-effects.

Do you have a preference?

g-ortuno, 2017-01-23:
I was trying to map this to the actual C++ code and it seemed weird to be clearing the map twice. I no longer think we should change anything.

closed 2017-01-23 by g-ortuno

g-ortuno, 2017-01-23:
Ah I actually missed the purpose of Step 1 when implementing disconenct: http://crbug.com/684073