whatwg/bluetooth #336: getPrimaryService is stuck

imlllin18, 2016-11-22:
I don't know what happened with getPrimaryService. The log from chrome stops after getPrimaryService. I check the log from the platform and chrome at the same time. However, the notify from device is received successfully based on the log from the platform. (see file btmon.btsnoop) 

I can't retrieve the received data if i don't get the service...
And I've read the many relevant issues but I still cannot solve it.

log from the platform:
[btmon.btsnoop.zip](https://github.com/WebBluetoothCG/web-bluetooth/files/605804/btmon.btsnoop.zip)
part of the code:
`navigator.bluetooth.requestDevice({filters: [{services: ['battery_service']}]})
	.then(device => {
		this.device = device;
		console.log('> Found ' + this.device.name);
		return device.gatt.connect();
	})
		.then(server => {
		console.log('Getting server...'+server.connected);
		this.server = server;
		
		console.log('server device: '+this.server.device);

		console.log('server device: '+this.server.device);
		return Promise.all([
			wait(3000),
			console.log('after 3 second '),		
			server.getPrimaryService(0x180f)
			.then(service => {
				console.log(service.uuid);
			  return this._cacheCharacteristic(service, 'battery_level');
			}),
		]);
	})`

I'm using ubuntu 16.04 LTS with google-chrome-unstable 56.0.2922.1 dev (64-bit)

beaufortfrancois, 2016-11-22:
FYI, using `Promise.all` will run `wait `and `server.getPrimaryService` at the same time. You may want to do something like this below.

```js
console.log('server device: '+this.server.device);
return wait(3000)
.then(_ => {
  console.log('after 3 second ');
})
.then(_=> this.server.getPrimaryService(0x180f))
.then(service => {
  console.log('service get');
  console.log(service.uuid);
  return this._cacheCharacteristic(service, 'battery_level');
})
.then(values => { console.log('value' + values); });

beaufortfrancois, 2016-11-22:
@imlllin18 Which BlueZ version did you use?
See https://github.com/WebBluetoothCG/web-bluetooth/blob/gh-pages/implementation-status.md#notes:

> Linux: Requires Kernel 3.19+ and BlueZ 5.41+ installed. Read How to get Chrome Web Bluetooth working on Linux.
> Note that Bluetooth daemon needs to run with experimental interfaces: `sudo /usr/sbin/bluetoothd -E`

beaufortfrancois, 2016-12-26:
@imlllin18 (gentle ping)

jyasskin, 2017-01-21:
I'm going to close this as something that would work better in https://crbug.com/, web-bluetooth@chromium.org, or irc://irc.w3.org/#web-bluetooth.

closed 2017-01-21 by jyasskin