whatwg/bluetooth #340: maplike handling is really weird

bzbarsky, 2016-12-12:
There are various parts of this spec that attempt to define what the "map entries" are for maplikes in terms of a Map.  But maplikes already have a Map whose [[MapData]] is defined to be the "map entries".  See https://heycam.github.io/webidl/#es-maplike

So whatever you're doing with `[[data]]` right now should instead be done by manipulating the `[[BackingMap]]`, I would think.  And you need to specify exactly how you manipulate it, of course.

jyasskin, 2017-01-20:
I don't know why I missed the `[[BackingMap]]` object that's available for this. Fix coming up.

jyasskin, 2017-01-20:
Oh, maybe I was trying to specify against the [language-independent maplike spec](https://heycam.github.io/webidl/#idl-maplike), which says that "Prose accompanying the interface can describe how the map entries of an object change.", instead of the [ES binding](https://heycam.github.io/webidl/#es-maplike), which provides `[[BackingMap]]`. I'll still send the change to use the ES-specific slots, since I'm assuming ES elsewhere.

closed 2017-01-24 by jyasskin

bzbarsky, 2017-01-25:
You skipped the "you need to specify exactly how you manipulate it, of course" bit.  Filed https://github.com/WebBluetoothCG/web-bluetooth/issues/351 on that, I guess...

PR #350 (2017-01-24): Use the [[BackingMap]] provided by the ES maplike binding.
Instead of a custom [[data]] slot that also holds an EcmaScript Map.

Fixes #340 

Preview at https://api.csswg.org/bikeshed/?url=https://raw.githubusercontent.com/jyasskin/web-bluetooth-1/use-backing-map/index.bs#dom-bluetoothadvertisingevent-bluetoothadvertisingevent
commit 2742a833 (2017-01-24): Use the [[BackingMap]] provided by the ES maplike binding. (#350)

Instead of a custom [[data]] slot that also holds an EcmaScript Map.

Fixes #340