whatwg/bluetooth #345: Adding Nordic UART Service

hewittwill, 2017-01-08:
Is there any plans to add the Nordic UART Service? 

The idea being is it'd allow communication to Arduino and other MCUs, and even allow Web based programming of them.

beaufortfrancois, 2017-01-09:
For info, Puck.js beacon already uses Web Bluetooth and Nordic UART Service.
See  https://www.puck-js.com/puck.js and https://www.kickstarter.com/projects/gfw/puckjs-the-ground-breaking-bluetooth-beacon

scheib, 2017-01-19:
@hewittwill, I'm not sure what you mean? Devices that use the nrf UART service expose it as a GATT service. You may use web bluetooth to acces it via the well known UUIDs listed at top of https://devzone.nordicsemi.com/documentation/nrf51/5.2.0/html/a00077.html

monteslu, 2017-01-19:
I'd like an official GATT service for `uart`, but there's no reason you couldn't just search by the service id.

jyasskin, 2017-01-19:
So far, we only [resolve names](https://webbluetoothcg.github.io/web-bluetooth/#resolveuuidname) for services that appear in https://www.bluetooth.com/specifications/gatt/services, and Nordic's UART service doesn't appear there. It'd be reasonable to define another way to resolve names, but I don't really want to be in the business of blessing particular companies' services, so I'd rather delegate to either some other organization or to the page. Do you have suggestions?

closed 2017-02-03 by jyasskin