whatwg/bluetooth #351: [[BackingMap]] manipulation is underdefined

bzbarsky, 2017-01-25:
The spec has things like:

>  Add a mapping from code to new DataView(bytes) in event.manufacturerData.[[BackingMap]]. 

What does that mean?  Is it doing `event.manufacturerData.[[BackingMap]].set(code, new DataView(bytes)` using whatever the current value of `%MapPrototype%.set` is?  Or is it using the default value of `%MapPrototype%.set`?  Or something else?

jyasskin, 2017-01-25:
Thanks for pushing. I'd like to be able to call out to https://infra.spec.whatwg.org/#map-set in order to manipulate the map entries. Does anything need to change in WebIDL to allow that?

@tobie

tobie, 2017-01-25:
As far as I'm concerned, you can tap into the infra standard for concepts and slots all you want.

jyasskin, 2017-01-25:
@tobie I'm worried that an [Infra map](https://infra.spec.whatwg.org/#maps) is not the same thing as an [ES Map](https://tc39.github.io/ecma262/#sec-map-objects), and WebIDL [says that [[BackingMap]] is an ES Map](https://heycam.github.io/webidl/#es-maplike).

Infra is careful to say that its lists are the same as ES's List type, and Infra's sets are specified as a particular kind of list, but Infra's maps are not specified as a kind of list. If they were, we could manipulate `[[BackingMap]].[[MapData]]` with the Infra algorithms. Is that the Right Thing to do? @annevk too, I guess.

tobie, 2017-01-25:
Oh, sorry. I didn't think that's what you were doing.

Manipulating ES types using infra terms sounds like a terrible idea.

bzbarsky, 2017-01-25:
Probably we should fix https://github.com/heycam/webidl/issues/254 and in the process have IDL expose some things other specs can use to do the guts of what IDL's generated `set` does...