whatwg/bluetooth #358: No popup to choose device

majaybabu, 2017-03-02:
Is it possible to directly connect to a peripheral without having the user choose the peripheral and click on 'pair' ?

g-ortuno, 2017-03-02:
No. Any time you want to connect to a device we will show the chooser. What use case did you have in mind?

majaybabu, 2017-03-02:
Reduce friction for user for e.g  when he is trying to make a payment on a merchant site and checkout. Connect to the device running GATT server with no 'device chooser popup' to get payment credentials. Want to remove friction around multiple clicks for the user on browser - 1. Click on a device 2. Click on Pair.

majaybabu, 2017-03-02:
Is this possible - Scan for devices and connect directly if there is only one device in the scan result. Without having the user choose that one device and pair - Can this feature be provided as an option ?

g-ortuno, 2017-03-02:
If we were to allowed that then any malicious site could connect any device around users.

thegecko, 2017-03-02:
There is of course the referring device from physical web

majaybabu, 2017-03-02:
Is it possible to customize the device chooser  - Change the text from 'wants to pair' to any other customized message and instead of 2 clicks on the chooser to just one click to start pairing.

majaybabu, 2017-03-03:
@thegecko 
Is it possible for the chrome running on OS X to scan for Eddystone URL beacon and then connect to  it using navigator.bluetooth.referringDevice ?  Any examples would really help.

jyasskin, 2017-03-16:
`referringDevice` isn't implemented yet, and when it is, that'll be user-directed scans, not page-directed scans. There will be a [scanning API](https://webbluetoothcg.github.io/web-bluetooth/scanning.html), but again, to communicate with a discovered device will take 2 permission prompts.

Chrome requires the two clicks from `requestDevice()` because it's pretty dangerous for users to grant access to a BT device, and we want to give them as much chance as possible to realize what they're doing. Even this level of concern still worries some people: http://www.tomshardware.com/news/google-web-bluetooth-webusb-apis,33589.html. However, another browser could certainly decide to use a different UI that lets the user select the first matching device with a single click.

closed 2017-03-16 by jyasskin

tigoe, 2018-08-10:
If we can't replace it with a custom chooser, would it be possible to make the device chooser conform to WCAG 2.0 guidelines? It doesn't currently work too well with a screen reader.

reillyeon, 2018-08-10:
@tigoe if you are having trouble using Chromium's chooser with a screen reader can you file a bug with details on crbug.com?

tigoe, 2018-08-10:
Will do. Though I assume web-bluetooth ever gets adopted by other browsers, this will be an API issue, not a browser issue, won't it? Ultimately developers will need to be able to customize all of the UI, including the chooser.

jyasskin, 2018-08-10:
As long as each browser's UI is accessible, accessibility won't require developers to customize it.

tigoe, 2018-08-10:
I haven't found the problem to be that tidy in my experience. Different
users have a wide range abilities and disabilities, and customization is
more often than not necessary. Disabilities don't scale well.

But I get your point. I'll seek a solution other than. Web-bluetooth.

VagrantZXC, 2020-05-22:
I saw the latest GATT Communication API list as:
[image]
And the status of Chooser UI is yes, does it mean we can customize the Chooser UI now?

reillyeon, 2020-05-22:
No, we have no plans to make the chooser UI customizable. The API status simply means that the chooser UI is implemented.

sbahmani, 2020-07-29:
Can I filter list with device id?