whatwg/bluetooth #359: Can't find my device - Mi Band 2

cpBurn, 2017-03-06:
I am trying to create a simple connection with my Mi Band 2 so I can read Heart Rate. I followed an example [here](https://webbluetoothcg.github.io/web-bluetooth/#example-heart-rate-monitor) but after requesting device, nothing appears on the list.

I also tried to use the example from the Demos folder. Using the [Heart Rate Sensor](https://webbluetoothcg.github.io/demos/heart-rate-sensor/) Demo I observed the same problem, the device doesn't appear at all.

The only way I could connect it was setting `acceptAllDevices: true`, like this:
`navigator.bluetooth.requestDevice({"acceptAllDevices":true, optionalServices: ['heart_rate']})`

In this way I can pair and connect the device, but when I try to `startNotifications()` I get following:

> DOMException: GATT operation failed for unknown reason.

Any ideas?

Tested on Chrome 56 and Chrome 59, MacOS 10.12.3
and Chrome Android 56.

g-ortuno, 2017-03-06:
Hi! Thanks for reporting problems. This repo is focused on the spec and since this appears to be a browser problem I opened an issue on Chrome's bug tracker: http://crbug.com/698852

I replied there with some more info.

closed 2017-03-06 by g-ortuno

cpBurn, 2017-03-06:
Awesome, thanks!

g-ortuno, 2017-03-07:
fyi: @beaufortfrancois added more information to http://crbug.com/698852