whatwg/bluetooth #362: requestDevice from extension

McMagnus, 2017-04-10:
I'm trying to write a chrome extension that should connect to a BLE device and use its services. However, when the web page posts the message to the extension, the user action (button click) is lost and when the extension calls requestDevice it fails because it doesn't think the call was initiated by a user action.

How is an extension supposed to connect to BLE devices?

I guess it might be possible to add a "onDeviceConnected" listener to the extension (when it's implemented), but will that work?

g-ortuno, 2017-04-10:
Extensions don't support Web Bluetooth. We talked about it in the past but there are currently no plans for support. Could you tell us about why you need an extension for this?

McMagnus, 2017-04-11:
Thanks for replying.

My answer is just one word: U2F

jyasskin, 2017-04-11:
Overall, this is a bug report for the Chrome implementation, rather than for the Web Bluetooth spec, so it should be filed at https://crbug.com/ rather than here. If you still think it's an issue, could you re-file there?

Also, the FIDO service, `fffd`, is [blocked](https://github.com/WebBluetoothCG/registries/blob/master/gatt_blocklist.txt#L28) because allowing arbitrary websites to access it defeats its security model. Allowing extensions to access it isn't much better, so I wouldn't expect an exception there.

closed 2017-04-11 by jyasskin