whatwg/bluetooth #363: s/gattserverdisconnected/gattserverdisconnect

beaufortfrancois, 2017-04-12:
From what I can see in several Web APIs such as WebUSB , Remote Playback and Presentation, they use `onconnect` and `ondisconnect`, not `onconnected` and `ondisconnected`, Therefore I believe we should follow that pattern and use `gattserverdisconnect` instead of `gattserverdisconnected`.

```idl
[ImplementedAs=ongattserverdisconnected] attribute EventHandler ongattserverdisconnect; 
```

g-ortuno, 2017-04-12:
fwiw I like gattserverconnected because it is very specific. Other APIs only have one type of connection Bluetooth has many types.

beaufortfrancois, 2017-04-12:
Just to be clear, I'm only arguing about the `ed` at the end that is present only in Web Bluetooth.

jyasskin, 2017-04-12:
Right, that's a sensible argument. It breaks compatibility with anyone currently using `gattserverdisconnected`, but we can fire both for a transition period.

@rbyers, do you have a sense of the right thing to do here? We probably shipped the wrong spelling of this event, but no other browsers have complained yet.

RByers, 2017-04-17:
The main risk here is that it's almost certainly better to just live with the mis-spelling then to try to fix it but fail (due to web compat) to remove the old name and so get stuck having both indefinitely.

Nobody else has implemented this as far as you know, right?  And usage is probably quite low?  Do you already have a UseCounter for WebBluetooth which would provide an upper-bound on the risk here?

Assuming we have evidence that the usage is pretty low then I'd suggest just making a concrete plan to deprecate/remove the old name by a specific milestone and ship the new event at the same time as deprecating the old one.  Hopefully you can get away with just firing both events during the transition period (rather than trying to do something complicated like we do for `mousewheel` where it's fired only when a listener for `wheel` wasn't invoked to prevent double handling).

Of course it's up to you guys whether the benefit of fixing the name is worth this hassle.  But as long as the usage is low hopefully the hassle/pain is really small...