whatwg/bluetooth #366: How to tell which device is the correct device when they all have the same name

whatsthebeef, 2017-05-06:
We currently use the chrome.bluetooth API on Chrome OS to discover our nearby BLE devices and at the same time retrieve RSSI information on these devices. This allows us to indicate to the user which is the closest device without the need for a pairing or a connection. 

Given that all our devices have the same name, there more than likely will be multiple devices operating within discoverable distance and the security restrictions with Web Bluetooth which, as I understand require the user to select (pair) the relevant devices from the list before viewing the RSSI information, how can the user know they are selecting the correct (closest) device without pairing with all devices.

My initial thought was that there could be a RSSI filter attribute on requestDevice on requestLEScan but given the current approach to security this may be a vulnerability as well.

This issue is more pertinent to us now given issues being raised against Chrome OS are being directed to Web Bluetooth as the new solution.

Apologies if I am missing something obvious. I assume this an issue which other products will have.

g-ortuno, 2017-05-06:
We recognized this use case as well so now Chrome shows an RSSI indicator next to each device in the Device chooser:

[image]

There is also the [Web Bluetooth Scanning API](https://webbluetoothcg.github.io/web-bluetooth/scanning.html), which hasn't been implemented yet, which would allow developers to receive devices' RSSI.

g-ortuno, 2017-05-08:
Closing since this seems to solve your problem. Feel free to re-open if it doesn't.

closed 2017-05-08 by g-ortuno

antoinegilles, 2018-07-10:
i open the subject, is there a method wich can calculate the RSSI in javascript ?

batuhanarslan, 2021-09-01:
The OS can recognize the RSSI before connecting to device and shows us the indicator. Is there any way to obtain this RSSI value before connecting to device aim to connect the true device?