whatwg/bluetooth #367: getPrimaryService() not resolving appropriately

NAllred91, 2017-05-07:
I'm experiencing issues with the getPrimaryService promise not resolving appropriately when refreshing the page and immediately trying to reconnect.

My user experience is the following:

- Make the initial connection to the device.
- Once the connection is completely finished, refresh the page.
- Immediately try to connect to the device again.
- May need to repeat this a few times, but eventually the connection process hangs.

In the code I'm observing that it is the `getPrimaryService` promise that simply never resolves.  I've even checked to make sure `device.gatt.connected` is true before making the call.

Once I get into this state the promise hangs indefinitely.  If I power off and power on my device and start the connection process in my browser again, without refreshing, the following happens:

- I make the call to `device.gatt.connect`
- Then, the initial call to `getPrimaryService` resolves with the error  "NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`."
- Then, the promise from `device.gatt.connect` resolves with the error "GATT Server disconnected while retrieving characteristics."
- At this point, I can attempt the connection again and it appears to always be successful.

I've tried to write some code to work around this edge case, (setting a timeout, telling the user to restart their device, etc...) but it seems like their might be some deeper issue here?

g-ortuno, 2017-05-07:
This seems to be a problem in Chrome's implementation of the API. Would you mind [filling an issue](https://bugs.chromium.org/p/chromium/issues/entry?components=Blink%3EBluetooth&source=chromium.org) for it in Chrome's bug tracker? Some logs would be really useful, instruction [here](https://www.chromium.org/developers/how-tos/file-web-bluetooth-bugs). What platform is this on?

NAllred91, 2017-05-08:
This is on a mac.  I'll try and get that filed tonight.  Thanks for pointing me in the right direction.

g-ortuno, 2017-05-08:
No problem! Closing since this appears to be a Chrome issue rather than a spec issue.

closed 2017-05-08 by g-ortuno