whatwg/bluetooth #370: errorSecurityError: Origin is not allowed to access the service. Tip: Add the service UUID to 'optionalServices' in requestDevice() options. https://goo.gl/HxfxSQ

brave2016, 2017-07-17:
function search(){
	document.getElementById("demo").innerHTML="My First JavaScript";
	let options = {};
	let filters = [];
	let filterName = document.querySelector('#name').value;
  if (filterName) {
    filters.push({name: filterName});
	 options.filters = filters;
  }else{
  options.acceptAllDevices = true;
  }
  
	navigator.bluetooth.requestDevice(options).then(device => {
			console.log('Got device:', device.name);
			console.log('id:', device.id);
			console.log('Connected:' + device.gatt.connected);
			console.log('Connected:' + device.gatt.connected);
			document.getElementById("demo").innerHTML="My First JavaScript"+device.name;
	return	device.gatt.connect();
		       
			}).then(server => {
				document.getElementById("service").innerHTML="trurservcice";
			
			 return server.getPrimaryService('00000000-0000-1000-8000-00805f9b34fb');
			}).then(service => {
			
			return Promise.all([
			
	              service.getCharacteristic('00000002-0000-1000-8000-00805f9b34fb'),
				  
				  service.getCharacteristic('00000003-0000-1000-8000-00805f9b34fb').then(read)
			
			]);
	
             })
			
			.catch(error => {
			document.getElementById("demo").innerHTML="error"+error;
			});
	
}

function read(characteristic){
    return characteristic.startNotifications()
    .then(char => {
	   characteristic.addEventListener('characteristicvaluechanged',
                                    onHeartRateChanged);
    });
}

function onHeartRateChanged(event) {
  let characteristic = event.target;
  console.log(parseHeartRate(characteristic.value));
  document.getElementById("data").innerHTML="D"+characteristic.value;
}

beaufortfrancois, 2017-07-18:
If you use `acceptAllDevices: true` and your `filters` don't include `00000000-0000-1000-8000-00805f9b34fb` you may want to use `optionalServices` as in https://googlechrome.github.io/samples/web-bluetooth/read-characteristic-value-changed.html

I guess something like this should work:

```js
  options.acceptAllDevices = true;
  options.optionalServices = ['00000000-0000-1000-8000-00805f9b34fb'];
```

closed 2017-07-18 by beaufortfrancois

brave2016, 2017-07-19:
On android 7.1.1, acceptAllDevices = true is invalid and must be written in filters, such as the filter device name, Services must be written to be able to run, and 6.0 will not be

beaufortfrancois, 2017-07-19:
Can you share the page you use @brave2016 as I'm not able to reproduce your issue?

brave2016, 2017-07-19:
https://github.com/a-kitchen/a-kitchen.github.io

beaufortfrancois, 2017-07-19:
You had a typo in `optionalServices`. See below:

```diff
function search() {
    document.getElementById("demo").innerHTML = "My First JavaScript";
    let options = {
-        optionalServices: ['00000003-0000-1000-8000-00805f9b34fb']
+        optionalServices: ['00000000-0000-1000-8000-00805f9b34fb']
    };
    let filters = [];
    let filterName = document.querySelector('#name').value;
    if (filterName) {
        filters.push({
            name: filterName
        });
        options.filters = filters;
    } else {
        options.acceptAllDevices = true;
    }

    navigator.bluetooth.requestDevice(options).then(device => {
        return device.gatt.connect();

    }).then(server => {
        document.getElementById("service").innerHTML = "trurservcice";

        return server.getPrimaryService('00000000-0000-1000-8000-00805f9b34fb');
    }).then(service => {

        return Promise.all([
            service.getCharacteristic('00000002-0000-1000-8000-00805f9b34fb'),
            service.getCharacteristic('00000003-0000-1000-8000-00805f9b34fb').then(read)
        ]);

    }).catch(error => {
        document.getElementById("demo").innerHTML = "error" + error;
    });
}
```

brave2016, 2017-07-19:
https://github.com/a-kitchen/a-kitchen.github.io

It's been added to 'optionalServices' or it's a mistake

    let options = {
    optionalServices:['00000000-0000-1000-8000-00805f9b34fb'],
    optionalServices:['00000001-0000-1000-8000-00805f9b34fb'],
    optionalServices:['00000002-0000-1000-8000-00805f9b34fb'],
    optionalServices:['00000003-0000-1000-8000-00805f9b34fb'],
    };

beaufortfrancois, 2017-07-19:
`optionalServices` should only include service UUIDs not characteristic UUIDs.

brave2016, 2017-07-27:
NotSupportedError: GATT operation failed for unknown reason. 

https://github.com/a-kitchen/a-kitchen.github.io/commits/master

 let chosenHeartRateService = null;

function search(){
    document.getElementById("demo").innerHTML="My First JavaScript";
    let options = {
    optionalServices:['00000000-0000-1000-8000-00805f9b34fb']
    };
    let filters = [];
    let filterName = document.querySelector('#name').value;
    if (filterName) {
    filters.push({name: filterName});
     options.filters = filters;
    }else{
     options.acceptAllDevices = true;
    }
  
    navigator.bluetooth.requestDevice(options).then(device => {
    return    device.gatt.connect();
    
               }).then(server => {        
             return server.getPrimaryService('00000000-0000-1000-8000-00805f9b34fb');
            }).then(service => {
             chosenHeartRateService = service;
            return Promise.all([
                  service.getCharacteristic('00000002-0000-1000-8000-00805f9b34fb').then(Notifications),
                  service. getCharacteristic('00000002-0000-1000-8000-00805f9b34fb').then(Notifications),
                  service.getCharacteristic('00000001-0000-1000-8000-00805f9b34fb').then(w),
                  service.getCharacteristic('00000003-0000-1000-8000-00805f9b34fb').then(read)
            ]);
    
             }).catch(error => {
            document.getElementById("demo").innerHTML=error;
            });
}

function Notifications(characteristic){
     characteristic.startNotifications();  
     
}

function read(characteristic){
   
    return characteristic.readValue().then(sensorLocationData => {
        
        if(sensorLocationData.zIndex >1){
        
        let sensorLocation = sensorLocationData.getUint8(1);
            document.getElementById("data").innerHTML="sensorLocation"+sensorLocation;
        }
        
    });
    
}

function w(characteristic){
    let resetEnergyExpended = new Uint8Array([34,8,35,7]);
    characteristic.writeValue(resetEnergyExpended);
    document.getElementById("w").innerHTML="write";
}