whatwg/bluetooth #372: No trigger characteristicvaluechanged method

brave2016, 2017-08-17:
There is no problem with the following code

 let chosenHeartRateService = null;
 

function search(){
	document.getElementById("demo").innerHTML="My First JavaScript";
	let options = {
		optionalServices:['00000000-0000-1000-8000-00805f9b34fb']
	};
	let filters = [];
	
	let  services = document.querySelector('#services').value;
	let  filterName = document.querySelector('#name').value;
	let  Prefix = document.querySelector('#Prefix').value;
	if(services || filterName || Prefix){
    if (filterName) {
    filters.push({name:filterName});
	}
	 if(services){
		 filters.push({services:[services]});
	 }
	 if(Prefix){
		  filters.push({namePrefix: Prefix});
	 }
	 options.filters = filters;
	 
	 
    }else{
     options.acceptAllDevices = true;
    }
  
	navigator.bluetooth.requestDevice(options).then(device => {
		
	return	device.gatt.connect();
	
		       }).then(server => {		
			 return server.getPrimaryService('00000000-0000-1000-8000-00805f9b34fb');
			}).then(service => {
			 chosenHeartRateService = service;
			return Promise.all([
			
				  service.getCharacteristic('00000001-0000-1000-8000-00805f9b34fb').then(w),
				  service.getCharacteristic('00000003-0000-1000-8000-00805f9b34fb').then(read)
			]);
	
             }).catch(error => {
			          document.getElementById("demo").innerHTML=error;
			});
}

function read(characteristic){
  return characteristic.startNotifications()
  .then(char => {
    characteristic.addEventListener('characteristicvaluechanged',
                                   onHeartRateChanged );
  });
	
}
//var tmp_count = 0 ;
function onHeartRateChanged (event) {
     var value = event.target.value;
  	document.getElementById("data").innerHTML="Received"+sensorLocation;
	//tmp_count ++ ;
		//document.getElementById("data").innerHTML=tmp_count;
}

function w(characteristic){
    let resetEnergyExpended = new Uint8Array([34,8,35,7]);
    characteristic.writeValue(resetEnergyExpended);
	document.getElementById("w").innerHTML="write";
}

beaufortfrancois, 2017-08-17:
Can you provide some logs?

brave2016, 2017-08-17:
I don't have a journal in my notepad 

 
&mdash;

brave2016, 2017-08-17:
An abnormality occurred while reading
  NotSupportedError: GATT Error Unknown.
NotSupportedError: GATT operation failed for unknown reason.

beaufortfrancois, 2017-08-17:
When do you get this error exactly?
It may be because your device is not Bluetooth Low Energy.

brave2016, 2017-08-17:
Bluetooth is low power. It was written from the start, and there was no problem with it

brave2016, 2017-08-17:
It's on Android devices

brave2016, 2017-08-18:
In opera, this property acceptAllDevices = true;ineffectiveness

beaufortfrancois, 2017-08-18:
Can you reproduce with Google Chrome?

brave2016, 2017-08-18:
That's the problem with android 7.1.1, and 6.0 won't be tested

beaufortfrancois, 2017-08-18:
Can you reproduce with Google Chrome, not Opera?

brave2016, 2017-08-18:
On Opera options. AcceptAllDevices = true;This property is invalid, and I'm now using Chrome for Google

beaufortfrancois, 2017-08-18:
Does it work fine for you now with Google Chrome?

nemeros111, 2020-01-22:
Sorry, Did you solved this problem?
My code working fine in PC chrome, but not in Android chrome.
Have some problem in here:
characteristic.startNotifications().then(_ => {
            characteristic.addEventListener('characteristicvaluechanged', handleNotifications);
          });

scheib, 2020-02-03:
@nemeros111 sounds like a Chrome on Android bug.  Can you help provide more detail on how to reproduce this, and logs: https://www.chromium.org/developers/how-tos/file-web-bluetooth-bugs

martpie, 2020-09-25:
Has anyone found a workaround to this issue? It seems like `characteristicvaluechanged` is never triggered on Android.