whatwg/bluetooth #375: it not showing while scanning printer device in Chrome android mobile

dineshvasan, 2017-08-21:
This is for sample example  ( https://dineshvasan.github.io/test/)
 i have written for connecting bluetooth device for printing the text but it is not working. kindly check it whether i have made mistake in that code . kindly give me working solution for web bluetooth using service worker. i have referred  all the website but it not working properly.

As per my requirement to access the web bluetooth print the text using service worker.
Kindly do the Needful 
Printer Name : Epson
Printer Model : Epson TM-P20 
`
[image]

[image]
`

beaufortfrancois, 2017-08-23:
Can you share some logs?
What do you see?

dineshvasan, 2017-08-24:
I have shared log file.kindly find the attachment .

Kindly give me working sample of web bluetooth

[btmon.btsnoop.tar.gz](https://github.com/WebBluetoothCG/web-bluetooth/files/1248195/btmon.btsnoop.tar.gz)

closed 2017-08-24 by beaufortfrancois

dineshvasan, 2017-08-24:
Its ok fine. i am searching on my mobile in chrome android os version 6.01  this is that site given below (https://googlechrome.github.io/samples/web-bluetooth/device-info.html?allDevices=true ) even not finding another mobile device in Android chrome browser what that exact problem ...

dineshvasan, 2017-08-24:
Kindly give me a working solution buddy and i am waiting for your response because i have

beaufortfrancois, 2017-08-24:
Here it is: https://groups.google.com/a/chromium.org/forum/#!topic/web-bluetooth/X26-_58ykVQ