whatwg/bluetooth #381: Simplified bluetooth beacon scanning API

jcampbell05, 2017-11-20:
Right now the API for specify you want to scan for a beacon is very complicated. I already have a UUID in string form for an iBeacon.

It would be nice to simplify pass in this uuid and perhaps some kind of identifier to specify this is a ibeacon uuid (as opposed to a eddystone one). 

Currently thats a lot of code just to search for a beacon

[image]

PrajualP, 2018-08-09:
Sir, is requestLEScan implemented in browsers? The implementation Status shows no.
Did you get the advertisement packets from beacons?

jcampbell05, 2018-08-09:
Not implemented as yet, this issue is based on the spec as was implementing in my work.