whatwg/bluetooth #382: Transfer URL between web applications

MirindaLucjo, 2018-01-02:
Hello,

Is there any way that a web application could leverage this web-bluetooth API to transfer an URL between another instance of the web application on another device? Both applications would have the same origin.
If my understanding of BLE is correct, than this would involve simulating a GATT service one of the devices, as it is done in the native application [ble-test-peripheral-android](https://github.com/WebBluetoothCG/ble-test-peripheral-android)?

jvdassen, 2018-01-03:
Hi,

@kenchris I just recently watched your [talk](https://www.youtube.com/watch?v=R8Vbv3SvAFA) at Mobcon where you said that it is only possible for web pages to interact with peripherals, is this still true?

jespertheend, 2021-09-12:
It seems like this would become possible with #231