whatwg/bluetooth #387: status of start/stop notification for chrome windows

dmazzella, 2018-02-14:
What about the status of implementation about startNotifications/stopNotifications for use event on characteristicvaluechanged in chrome for windows?

Where i can read about this?

Thanks and regards,
Damiano.

beaufortfrancois, 2018-02-15:
@cco3 may be able to help there.

cco3, 2018-02-16:
At the device level, subscribe has an implementation (I've not tested it), but unsubscribe does not.
crbug.com/735828
However, this may be covered up by the web bluetooth layer (i.e., maybe unsubscribe functionally works there).  I'd have to do more testing to see.

dmazzella, 2018-02-16:
@cco3, I am using for testing the latest version of Chrome version 66.0.3349.0 (official Build) Canary (64 bit).

A quick way to verify that the expected events from the "characteristicvaluechanged" listener are not received:

1) Open the application for iOS: LightBlue
2) Create a virtual peripheral: Heart Rate
3) Select the peripheral you just created for enable it: Heart Rate

1) Pair the newly created peripheral: Windows-> Settings-> Bluetooth-> pair a device-> Heart Rate-> Insert pin.
2) Open Chrome Canary at URL: chrome://flags/#enable-Experimental-web-platform-features-> enable (if disabled)
3) Restart Chrome Canary to apply the changes (if point 2 was disabled)
4) Open Chrome Canary at URL: https://googlechrome.github.io/samples/web-bluetooth/notifications.html
5) Select Bluetooth Service: heart_rate
6) Select Bluetooth Characteristic: heart_rate_measurement
7) Start Notifications

The output is:

Live Output
```
Requesting Bluetooth Device...
Connecting to GATT Server...
Getting Service
Getting Characteristic...
> notifications started
```
And nothing else.

From Chrome with the Bluetooth Web extension polyfill (Web Bluetooth polyfill 0.4.0-Windows 10 web Bluetooth API polyfill)
The output is:

Live Output
```
Requesting Bluetooth Device...
Connecting to GATT Server...
Getting Service
Getting Characteristic...
> notifications started
> 0x00 0x78
> 0x00 0x78
> 0x00 0x78
> 0x00 0x78
> 0x00 0x64
> 0x00 0x64
> 0x00 0x64
> 0x00 0x50
> 0x00 0x50
> 0x00 0x50
> notifications Stopped

```
Can you give me some feedback?

Best regards and thanks, 
Damiano

cco3, 2018-02-16:
Thanks for testing!  I'm guessing that the notification subscription as it exists in Chrome on Windows was probably implemented and working for some other feature in Chrome.  At the moment, I'm not sure what it will take to get it working for Web Bluetooth.

dmazzella, 2018-02-19:
> At the moment, I'm not sure what it will take to get it working for Web Bluetooth.

when do you think you can take a look?

dmazzella, 2018-03-02:
news about this?

brianz, 2018-03-02:
I'm also very interested in this. I'm currently developing a Web Bluetooth app with a custom device which sends notifications. Having native support for Windows would be a huge 👍 . I'd be more than willing to do some testing.

cco3, 2018-03-02:
Sorry, planning on taking another look soon.

vshymanskyy, 2018-03-28:
Works fine on all platforms except... Windows! :)

jsiebert, 2018-04-18:
Are there any news on this issue? I'm currently writing an article about the Espruino Platform and wanted to show how to use Google Chrome and Web Bluetooth for their IDE. Doesn't work though...

cco3, 2018-05-14:
Sorry, I don't have any news here and I'm no longer actively working on Web Bluetooth, but the web bluetooth team is still aware of the issue and we are still pursuing Windows support.

motivado, 2018-05-28:
Hi, are there any news on the topic?

vshymanskyy, 2018-05-28:
This is so annoying as it seems to be a minor addition, but restricts so many applications working on a major OS... Which holds back the development of creative IoT services :(

dmazzella, 2018-08-20:
Solved. now work on windows!