whatwg/bluetooth #388: Can not detect Bluetooth with requestDevice.

goldingdamien, 2018-03-01:
I have tried navigator.bluetooth.requestDevice(options) with options = {acceptAllDevices: true} and using the name in filters against a new Bluetooth phone(Bluetooth 4+) and an older device which both are fine at pairing with my computer. 

I have tried following the guides for a solution but everything seems to result in the scanning overlay being shown and it never finding anything. 

I have tried using Chrome Canary, an onclick event, and using https. 

I have also tried the below, and nothing is shown.
chrome://bluetooth-internals/#devices

I just want to be able to test the Bluetooth functionality, so any way of getting this working would be much appreciated, thank you.

GraxMonzo, 2018-10-18:
Same issue

Guziq, 2018-10-18:
Interesting, now I have to enable `chrome://flags/#enable-experimental-web-platform-features` on MacOS (that wasn't a case previously).

reillyeon, 2018-11-21:
Closing this issue as OP has not provided information on what platform they are using. Note that as of Chrome 70 Windows is now fully supported and so this may have fixed your problem.

closed 2018-11-21 by reillyeon

piyush2cloud, 2020-12-02:
@reillyeon Can u please help me with how to get heart rate from Fitbit Versa device using BLE,GATT.