whatwg/bluetooth #393: Cannot interact with non-GUI HID interface

jhaenchen, 2018-04-25:
Is there any new info about how I should use HID devices over Bluetooth? It seems oh so very reasonable to use the HID protocol to do a million different things, many of which have nothing to do with audio streams or keyboard inputs.

chrome.hid won't allow you to interact with HID pages associated with keyboard or mouse input, but they will allow everything else. Can we allow a similar filtering system for web bluetooth?

Totally blacklisting the HID protocol seems like overkill to me. A softphone is an easy example of where this functionality would be useful.

g-ortuno, 2018-04-27:
IIRC, Android blocks access to HID GATT because of security concerns. I think that validates Web Bluetooth's decision to also block access to it.

g-ortuno, 2018-04-27:
A Web API that specifically targets HID and takes appropriate security measures seems like a better alternative.

closed 2018-04-27 by g-ortuno

jracle, 2018-04-27:
Hi @g-ortuno ,

If I can add my voice here, our [Logitech Unifying for Chrome](https://chrome.google.com/webstore/detail/logitech-unifying-for-chr/agpmgihmmmfkbhckmciedmhincdggomo) app uses [chrome.hid](https://developer.chrome.com/apps/hid) API we co-developed to communicated with our unifying receiver using HID vendor. It would be great indeed to allow readonly access to top-level collections like consumer page, and read-write access to vendor page.

I'm pessimistic in the probability of success of standardization of a W3C version (say web-hid), considering the fact that we are not even done pusing web-bluetooth to Edge and Safari team! If we Logitech can help further, of course we'll do!

hilowshiy, 2018-04-27:
This is not my account.  If it is under my info please delete it..

jracle, 2018-04-27:
@hilowshiy ?

jhaenchen, 2018-04-27:
@g-ortuno I think it's also interesting to see where WebUSB decided to allow HID at a higher level interface than the typical USB https://github.com/WICG/webusb/issues/29

This is a very similar situation to WebBluetooth's.

jhaenchen, 2018-04-30:
@reillyeon As I understand it, Chrome did go with your approach #1: "Implement new WebUSBHID interface for Chrome and provide to Chromium as spec/patch. It might be like chrome.hid on top of usbhid hid user space library." chrome.hid does just that, and I have used chrome.hid to interact with some telephony functionalities, though I can't say for sure i was using telephony usage pages to do so.

The problem is that chrome.HID is built on the assumption that you're interacting with the device over USB. I suppose in a perfect world chrome.HID would be an interface that could interact with any authorized device be it USB or Bluetooth GATT. As of now there seems to be no way to interact with HID over bluetooth, and I see no way for chrome.HID to be such a solution without an implementation by the WebBluetooth team.

I would consider it a very good design indeed for these protocols to play nice with each other, allowing us to build technologies that use devices from a variety of sources. These are not, after all, fundamentally different use cases between USB and Bluetooth, just different means of connecting you to your device.

jhaenchen, 2018-04-30:
PS is there any way this could remain open? It seems clear to me that this issue is not quite settled.

reillyeon, 2018-04-30:
chrome.hid is not a web API since it is restricted to Chrome Apps and predates WebUSB by years so it isn't an implementation of the first approach [there](https://github.com/WICG/webusb/issues/29#issuecomment-274623755). To my knowledge there has been no movement on the second approach. The third approach is what I recommend to developers building devices they intend on using with WebUSB.

It's unfortunate that the work to make chrome.hid work reliably with Bluetooth devices was never done.

To be clear, what would need to be done to Web Bluetooth to unblock the HID GATT service is to implement support for parsing the HID messages encoded in GATT messages and applying similar filtering to what is done in the chrome.hid API. This behavior would need to be carefully specified so that it could be implemented consistently across browsers.

jhaenchen, 2018-05-03:
@reillyeon @g-ortuno I honestly think it would be worthwhile. How can a general bluetooth API block one of the most common protocols for interacting with devices, especially the only one shared with USB? Come on folks

s-light, 2019-06-12:
i have no idea what status this Idea/Repository is in - but i found it and would be the 'solution'?
- [WebHID (Human Interface Device)](https://www.chromestatus.com/features/5172464636133376)
- [WebHID Explainer](https://github.com/WICG/webhid/blob/master/EXPLAINER.md)