whatwg/bluetooth #394: ENABLE_NOTIFICATION_VALUE - Web Bluetooth API

PaulKotlov, 2018-05-02:
Client Configuration Characteristic Descriptor (CCCD).
UUID - "00002902-0000-1000-8000-00805f9b34fb" / 'gatt.client_characteristic_configuration'.

To set CCCD in Java code for Android
we do:

`    public static final byte[] ENABLE_NOTIFICATION_VALUE = {0x01, 0x00};
    
    BluetoothGattDescriptor descriptor =    characteristic.getDescriptor("00002902-0000-1000-8000-00805f9b34fb");
    descriptor.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
    mBluetoothGatt.writeDescriptor(descriptor);
`

How can I perform a similar configuration in JavaScript code when using the Web Bluetooth API ?

My version in JavaScript:  
`    .then(descriptors => {    
    let queue = Promise.resolve();
     descriptors.forEach(descriptor => {
            switch (descriptor.uuid) {    
              case BluetoothUUID.getDescriptor('gatt.client_characteristic_configuration'):
               queue = queue.then(_ => descriptor.readValue()).then(value => {
                  descriptorCache = descriptor;
                });
       ...
    
        var data = new Uint8Array([0x01, 0x00]);
        descriptorCache.writeValue(data);
        //descriptorCache.writeValue(new TextEncoder().encode(data));
`         

fails with a security error :-(
 
Uncaught (in promise) DOMException: writeValue() called on blocklisted object marked exclude-writes. 
https://webbluetoothcg.github.io/web-bluetooth/#attacks-on-devices

I understand the need for security.
But after all, a lot of devices require a presetting of the CCCD.

If I don't understand correctly then please forgive me. 
I didn't find another way to resolve my issue. 

Also this question is available here: https://stackoverflow.com/q/50097579/9720482

jyasskin, 2018-05-02:
Calling [`characteristic.StartNotifications()`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothremotegattcharacteristic-startnotifications) sets the CCCD for you.

closed 2018-05-02 by jyasskin

PaulKotlov, 2018-05-04:
Jeffrey, thank You very much!