whatwg/bluetooth #397: can't pair with a device that require pin code

merylb, 2018-05-23:
hello,
till now i can discover my bluetooth device without problem, but i can't pair with this device cause it require  a pin code how can i pass this pin code using web-bluetooth api ?
please i want know if its possible or not

reillyeon, 2018-05-25:
The specification contains the following in the steps for [BluetoothRemoteGATTServer.connect()](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothremotegattserver-connect),

> Note: We would normally prefer to give the website control over whether and when bonding happens, but the Core Bluetooth platform API doesn’t provide a way for UAs to implement such a knob. Having a bond is more secure than not having one, so this specification allows the UA to opportunistically create one on platforms where that’s possible. This may cause a user-visible pairing dialog to appear when a connection is created, instead of when a restricted characteristic is accessed.

This implies that if the device requires pairing using a PIN code then the user will be prompted for this code when the site attempts to connect to the device.

AndriyHromyak, 2019-04-08:
Hi all, 
I am now working with web-bluetooth and my device is showing pin to pair, however browser is not prompting anything. Is this expected ? 
And because i need to fetch data stored in device when i am writing to descriptor 00002902-0000-1000-8000-00805f9b34fb it fails with error: writeValue() called on blocklisted object marked exclude-writes. maybe i am doing something wrong but my logic is that maybe because i am paired without using pin it is not allowing me to write to descriptor anything, im new to this so myabe my logic is wrong.

reillyeon, 2019-04-08:
Closing this as not a spec issue. I will answer this question on StackOverflow.

closed 2019-04-08 by reillyeon