whatwg/bluetooth #403: Why display “Connected: false” when pair to BLE device in Web Bluetooth API sample?

n1705771, 2018-07-09:
I open this [Web Blutooth API sample](https://googlechrome.github.io/samples/web-bluetooth/device-info-async-await.html) in my Android phone:

I select "All Devices" and click the button "Get Bluetooth Device Info", then my HM-10 BLE device is shown. I select this device and click "pair" to connect the device, but Live Output shows

connected: false

To my understanding, the possible reason causes the problem is this web bluetooth api doesn't set correct UUID for HM-10 BLE module. But I don't know where to set the UUID.

Could you please help?

beaufortfrancois, 2018-07-09:
I understand why it can be confusing. To actually connect to the device, you have to call `await device.gatt.connect()`. See https://googlechrome.github.io/samples/web-bluetooth/battery-level.html

I may need to update `Connected: false` to `GATT Connected: false`. Would that help? 

See https://developers.google.com/web/updates/2015/07/interact-with-ble-devices-on-the-web#connect_to_a_bluetooth_device

n1705771, 2018-07-09:
Can I directly test it using that web page: https://googlechrome.github.io/samples/web-bluetooth/battery-level.html ?
I open the page in my phone and PC, then click the button. But seems it cannot find any devices...

Thanks

beaufortfrancois, 2018-07-10:
As recommended at https://developers.google.com/web/updates/2015/07/interact-with-ble-devices-on-the-web#samples_demos_and_codelabs, you can install the [BLE Peripheral Simulator Android App](https://play.google.com/store/apps/details?id=io.github.webbluetoothcg.bletestperipheral) which simulates a BLE peripheral with a Battery Service, a Heart Rate Service, or a Health Thermometer Service.

n1705771, 2018-07-11:
Thanks! It works well!

closed 2018-07-11 by n1705771