whatwg/bluetooth #404: Compile error when integrate in my project

n1705771, 2018-07-11:
I am working on a Angular project. I created a component and insert the code bluetooth sample into the component ts file. The ts code is:

`import { Component, OnInit } from '@angular/core';

@Component({
  selector: 'app-bluetooth',
  templateUrl: './bluetooth.component.html',
  styleUrls: ['./bluetooth.component.css']
})
export class BluetoothComponent implements OnInit {

  constructor() {
  }

  ngOnInit() {
  }

  log() {
    var line = Array.prototype.slice.call(arguments).map(function(argument) {
      return typeof argument === 'string' ? argument : JSON.stringify(argument);
    }).join(' ');
    document.querySelector('#log').textContent += line + '\n';
  }

  clearLog() {
    document.querySelector('#log').textContent = '';
  }

  setStatus(status) {
    document.querySelector('#status').textContent = status;
  }

  setContent(newContent) {
    var content = document.querySelector('#content');
    while(content.hasChildNodes()) {
      content.removeChild(content.lastChild);
    }
    content.appendChild(newContent);
  }

  onButtonClick() {
    this.clearLog();
    this.log('Requesting any Bluetooth Device...');
    navigator.bluetooth.requestDevice({
     // filters: [...] <- Prefer filters to save energy & show relevant devices.
        acceptAllDevices: true,
        optionalServices: ['device_information']})
    .then(device => {
      this.log('Connecting to GATT Server...');
      return device.gatt.connect();
    })
    .then(server => {
      this.log('Getting Device Information Service...');
      return server.getPrimaryService('device_information');
    })
    .then(service => {
      this.log('Getting Device Information Characteristics...');
      return service.getCharacteristics();
    })
    .then(characteristics => {
      let queue = Promise.resolve();
      let decoder = new TextDecoder('utf-8');
      characteristics.forEach(characteristic => {
        switch (characteristic.uuid) {

          case BluetoothUUID.getCharacteristic('manufacturer_name_string'):
            queue = queue.then(_ => characteristic.readValue()).then(value => {
              this.log('> Manufacturer Name String: ' + decoder.decode(value));
            });
            break;

          case BluetoothUUID.getCharacteristic('model_number_string'):
            queue = queue.then(_ => characteristic.readValue()).then(value => {
              this.log('> Model Number String: ' + decoder.decode(value));
            });
            break;

          case BluetoothUUID.getCharacteristic('hardware_revision_string'):
            queue = queue.then(_ => characteristic.readValue()).then(value => {
              this.log('> Hardware Revision String: ' + decoder.decode(value));
            });
            break;

          case BluetoothUUID.getCharacteristic('firmware_revision_string'):
            queue = queue.then(_ => characteristic.readValue()).then(value => {
              this.log('> Firmware Revision String: ' + decoder.decode(value));
            });
            break;

          case BluetoothUUID.getCharacteristic('software_revision_string'):
            queue = queue.then(_ => characteristic.readValue()).then(value => {
              this.log('> Software Revision String: ' + decoder.decode(value));
            });
            break;

          case BluetoothUUID.getCharacteristic('system_id'):
            queue = queue.then(_ => characteristic.readValue()).then(value => {
              this.log('> System ID: ');
              this.log('  > Manufacturer Identifier: ' +
                  this.padHex(value.getUint8(4)) + this.padHex(value.getUint8(3)) +
                  this.padHex(value.getUint8(2)) + this.padHex(value.getUint8(1)) +
                  this.padHex(value.getUint8(0)));
              this.log('  > Organizationally Unique Identifier: ' +
                  this.padHex(value.getUint8(7)) + this.padHex(value.getUint8(6)) +
                  this.padHex(value.getUint8(5)));
            });
            break;

          case BluetoothUUID.getCharacteristic('ieee_11073-20601_regulatory_certification_data_list'):
            queue = queue.then(_ => characteristic.readValue()).then(value => {
              this.log('> IEEE 11073-20601 Regulatory Certification Data List: ' +
                  decoder.decode(value));
            });
            break;

          case BluetoothUUID.getCharacteristic('pnp_id'):
            queue = queue.then(_ => characteristic.readValue()).then(value => {
              this.log('> PnP ID:');
              this.log('  > Vendor ID Source: ' +
                  (value.getUint8(0) === 1 ? 'Bluetooth' : 'USB'));
              if (value.getUint8(0) === 1) {
                this.log('  > Vendor ID: ' +
                    (value.getUint8(1) | value.getUint8(2) << 8));
              } else {
                this.log('  > Vendor ID: ' +
                    getUsbVendorName(value.getUint8(1) | value.getUint8(2) << 8));
              }
              this.log('  > Product ID: ' +
                  (value.getUint8(3) | value.getUint8(4) << 8));
              this.log('  > Product Version: ' +
                  (value.getUint8(5) | value.getUint8(6) << 8));
            });
            break;

          default: this.log('> Unknown Characteristic: ' + characteristic.uuid);
        }
      });
      return queue;
    })
    .catch(error => {
      this.log('Argh! ' + error);
    });
  }

  /* Utils */

  padHex(value) {
    return ('00' + value.toString(16).toUpperCase()).slice(-2);
  }

  getUsbVendorName(value) {
    // Check out page source to see what valueToUsbVendorName object is.
    return value +
        (value in valueToUsbVendorName ? ' (' + valueToUsbVendorName[value] + ')' : '');
  }
}
`

However I got below compile errors. Seems I 

ERROR in src/app/bluetooth/bluetooth.component.ts(41,5): error TS2554: Expected 0 arguments, but got 1. 
src/app/bluetooth/bluetooth.component.ts(42,15): error TS2339: Property 'bluetooth' does not exist on type 'Navigator'.
src/app/bluetooth/bluetooth.component.ts(60,25): error TS2304: Cannot find name 'TextDecoder'.
src/app/bluetooth/bluetooth.component.ts(64,16): error TS2304: Cannot find name 'BluetoothUUID'.
src/app/bluetooth/bluetooth.component.ts(124,21): error TS2304: Cannot find name 'getUsbVendorName'.
src/app/bluetooth/bluetooth.component.ts(152,19): error TS2304: Cannot find name 'valueToUsbVendorName'.

beaufortfrancois, 2018-07-11:
I think @manekinekko will be able to help.

n1705771, 2018-07-11:
Very sorry! I feel I raised this issue in wrong area. I have raised it in another place and will close it here.

manekinekko, 2018-07-13:
@n1705771 let's move the discussion here https://github.com/manekinekko/angular-web-bluetooth/issues/42

Could you close this issue plz?

closed 2018-07-13 by g-ortuno