whatwg/bluetooth #406: "This feature is non-standard and is not on a standards track". Is this accurate?

timoxley, 2018-07-25:
The [`Bluetooth` page on MDN ](https://developer.mozilla.org/en-US/docs/Web/API/Bluetooth) says:

> This feature is non-standard and is not on a standards track

[image]

Is this (still) accurate? It appears the warning should perhaps be upgraded to [Experimental](https://developer.mozilla.org/en-US/docs/MDN/Contribute/Guidelines/Conventions_definitions#Experimental)?

[image]

beaufortfrancois, 2018-07-25:
@jpmedley may know what is the appropriate wording there.

beaufortfrancois, 2018-07-25:
Thank you @timoxley for raising this issue.

ExE-Boss, 2018-08-23:
I’ve done this as part of mdn/browser-compat-data#2609 and mdn/browser-compat-data#2611.

jpmedley, 2018-08-23:
The latter is correct. Thanks for fixing.

timoxley, 2018-08-27:
I believe this can be closed, the status messages on all the MDN pages I checked appear to be up to date.

closed 2018-08-27 by timoxley