whatwg/bluetooth #408: Observing characteristic notifications start and stop

rajsite, 2018-09-04:
The startNotifications and stopNotifications effect all event listeners for a characteristic.

I have an event listener that closes around some state for buffering messages that come in. When notifications stop I want to be able to clean-up that state which involves resolving some pending promises.

As is some external factor can call stopNotifications or startNotifications and I don't see a native way to observe that change.

Some ideas:
1. Add event listeners for gatt characteristic value change notifications starting and stopping
2. Implement `characteristic.getNotifier` returns a reference to a GATTCharacteristicNotifier object that I use for registering event listeners instead of registering on the characteristic. I can start and stop notifications on the GATTCharactertisticNotifier instance without impacting other observers (and them impacting me)
3. Maybe not have startNotifications and stopNotifications? There is no equivalent parallel in the DOM space for disabling the event system. Seems like this idea has been previously rejected: https://github.com/WebBluetoothCG/web-bluetooth/issues/236

g-ortuno, 2018-09-04:
What do you mean some external factor? After startNotifications() is called by a site, the site should receive notifications until stopNotifications() is called. The site should never receive a notification before startNotifications() or after stopNotifications(). In theory, nothing besides a disconnection should break that contract.

rajsite, 2018-10-14:
Been looking at this again more recently and I'll step back and provide more context. I'm trying to wrap the Web Bluetooth api to make it available in a custom language runtime compiled to WebAssembly.

In the spec it mentions that after startNotifications event handlers can be setup within the microtask checkpoint without losing messages. Unfortunately I can't make that guarantee for nodes that execute in our runtime (two nodes may run synchronously or as separate tasks based on how they are scheduled, can't predict).

So if I try and keep a thin abstraction and have startNotifications and register event listeners as separate nodes then users may miss notifications.

Best option I can think of is either always startNotifications and stopNotifications internally and don't expose that to users or make the nodes have a parameter on the event registration node and let users manage start / stop notifications while adding or removing event listeners.

What it boils down to is why does start and stop notifications exist? Is it only because it is weird for adding event listeners to perform the action of starting notifications? Is there any actual value of having those from the bluetooth device api perspective that I'm missing?

reillyeon, 2018-10-29:
The ability to register multiple event listeners is inherited from the rest of the web platform. The reason `startNotifications()` and `stopNotifications()` are critical for Web Bluetooth is that events can bubble up the [Bluetooth tree](https://webbluetoothcg.github.io/web-bluetooth/#bluetooth-tree) from the `BluetoothRemoteGattCharacteristic` they are fired against (though this is currently unimplemented in Chrome). Starting and stopping notifications implicitly based on whether an event listener was registered would not provide the information necessary to know which characteristic to start notifications on since the event listener could be registered higher up the tree.

rajsite, 2018-10-29:
Do you know what use cases are there for the Bluetooth tree? The same doesn't exist for the WebSockets api (ie having a globally accessible way to get at all existing connection instances).

It also brings the original concern back a bit. How can I know if some other script will be starting and stopping notifications on a characteristic I'm interacting with? I suppose DOM Events can have arbitrary scripts stopPropagation, etc. but I think of web bluetooth as more of a communications api where I want more control of messages and responses.

reillyeon, 2018-10-29:
I am not the original author of that part of the spec (and it isn't even implemented) but it seems reasonable that a site author may want to register a single event listener on a `BluetoothDevice` rather than individual listeners on each characteristic.

Other script on the page modifying a `BluetoothRemoteGattCharacteristic`'s state is certainly a concern however at that point I would generally be concerned that you have two pieces of script that are not coordinating in how they are communicating with the device in general.