whatwg/bluetooth #410: Origin is not allowed to access any service.

safeimuslim, 2018-09-28:
I got error like this, anyone can help?

`Uncaught (in promise) DOMException: Origin is not allowed to access any service. Tip: Add the service UUID to 'optionalServices' in requestDevice() options.`

My code like this
```
/* eslint-disable */
      const device = await window.navigator.bluetooth.requestDevice({
        acceptAllDevices: true,
        optionalService: ['000018f0-0000-1000-8000-00805f9b34fb']
      })
      console.log('DEVICE INFO: ', device.name);
      const server = await device.gatt.connect();
      console.log('server: ', server);
      const service = await server.getPrimaryService('000018f0-0000-1000-8000-00805f9b34fb');
      console.log('service: ', service);
      const characteristic = await service.getCharacteristic("00002af1-0000-1000-8000-00805f9b34fb");
      console.log('service: ', service);
```

beaufortfrancois, 2018-09-28:
It looks like the bluetooth device you're trying to doesn't have a service with UUID `000018f0-0000-1000-8000-00805f9b34fb`.

safeimuslim, 2018-09-28:
@beaufortfrancois How to get my device UUID ?

beaufortfrancois, 2018-09-28:
I would install https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en on your Android device, scan and connect to your Bluetooth device to inspect what is available.
I would recommend having a look at https://medium.com/@urish/reverse-engineering-a-bluetooth-lightbulb-56580fcb7546

closed 2018-09-28 by reillyeon

safeimuslim, 2018-09-29:
What diff about UUID PrimaryService and UUID Characteristic
```
        .then(server => server.getPrimaryService("000018f0-0000-1000-8000-00805f9b34fb"))
        .then(service => service.getCharacteristic("00002af1-0000-1000-8000-00805f9b34fb"))
```

j3rryzh, 2022-03-09:
You were missing an "s" in `optionalService`, should be `optionalServices`.

helentsai17, 2023-09-13:
I came up with the same issue and this helped me
https://developer.mozilla.org/en-US/docs/Web/API/BluetoothCharacteristicProperties
might need to use EventListener and startNotifications