whatwg/bluetooth #422: Service Worker support for WebBluetooth

hardillb, 2018-12-16:
An issue to track the adding ServiceWorker support to WebBluetooth.

Things to think about:

 - Could/Should this allow persistent connections to devices e.g. a User initiates connection to a device on phone/laptop, the bluetoothDevice object is passed to the ServiceWorker. The phone/laptop goes out of range, comes back later, can the ServiceWorker reconnect in the background? 
 -Should it ask permission again every time or just once with option to allow again until revoked?

beaufortfrancois, 2018-12-17:
@reillyeon @dougt for thoughts.

PavelCibulka, 2019-03-19:
Hi that would be nice. We use website as store application and connect barcode scanners as keyboard right now. 

We are experimenting with WebBluetooth on barcode scanner and weight scales (for parcels). Problem is that you have to reconnect device each page refresh / GET / POST. 

Each employee has his tablet and scanner. I think service worker with persistent bluetooth connections would really help.

reillyeon, 2019-03-19:
@PavelCibulka as an interim step Chrome is working on implementing support for retaining access to a device across navigations. That is still not quite as convenient as moving the connection management into a Service Worker but should help resolve the issue you are having. Follow along on [issue 577953](https://bugs.chromium.org/p/chromium/issues/detail?id=577953) for progress.

thegecko, 2019-03-19:
Could they also look at implementing `referringDevice` to close the loop on eddystone beacons connecting a user to a web-based UI?

gfwilliams, 2019-11-19:
Just to add my thoughts to this - I'm currently working on [a hackable smartwatch](https://banglejs.com/) - allowing users to keep a connection open in the background and communicate with their device would be a huge benefit.

odejesush, 2019-11-19:
**TL;DR: Web Bluetooth support in service workers is complicated and has many questions that need to be answered before it can be considered. Discussion on these questions is welcome to generate ideas for how to address them.**

There are some questions that need to be answered before Web Bluetooth support is able to be considered in service workers. These are just some that that pop into my head at the moment:
1. How should permissions look for Web Bluetooth in service workers?
2. Should all Bluetooth devices be allowed in a service worker context?
3. How should users be notified when a Bluetooth device is being accessed in the background?
4. Does having a connection to a GATT server count as not being idle, or only when I/O operations are occurring?

I do have some ideas to address or partially address some of these questions. It would be great to hear any thoughts on these or other ideas or questions that should be asked.

1. Allowing site to be able to access hardware in the background without the user's knowledge has some security and privacy concerns. A malicious site can either flash a device at any time, track a user's location by detecting whether paired devices are in range or not, or just read sensitive information, such as heart rate data in the background. There are two ideas that I have to address some of these concerns. The first is to gate background access behind a permission to allow background access that is in addition to the permission to connect to a device. This would be a signal from the user that they trust the site to communicate with the device. The second is to only allow background access in installed PWAs, as the user has to explicitly install the PWA themselves. It can even be combined with the first idea and only allow background access after the user as explicitly granted background access permission. This would provide two gates, so a user is less likely to be tricked by a malicious site.

2. There are devices where it doesn't make sense for background access to be given, such as piloting a drone. The question is, how can we detect whether a device should be allowed or not, or is it OK to allow all and let the user decide which devices can be accessed in the background via the permission prompts.

3. Users can be notified with a notification pop up when a device is being connected, but that could potentially become spammy and annoying to users if it occurs very frequently. A more subtle approach can be taken, such as a taskbar indicator or an indicator on the browser window, but then this can be easy to miss.

4. It makes sense to keep the service worker active when a device is actively connected. A malicious site could take advantage of this and open up a connection to do unrelated work in the background forever, like bitcoin mining for example. Even if there is a timeout implemented, sites could use the timeout period to perform unrelated computations. It also makes sense to keep the service worker active when I/O operations are being performed, but a malicious site can still perform dummy operations to do other work in the background.

Again, these are just some ideas that I have to address some of these concerns, and I welcome any other thoughts on them. Most of these questions also apply to WebUSB, Web Serial, and WebHID support in service workers.

antonbasic, 2019-11-20:
I have a use case for being able to update my BT device over BT while it's not in active use. This would also be perfect to do in a service worker.

In terms of how this should be done. I think there are quite good practice to be found on phones already. You get an initial question of whether or not you want to give the permission. Then after a while you get a notification that says that the app has been using your BT/GPS/whatever behind the scenes and whether or not you want to keep allowing that. From that point on ( I guess with some time limit ) you won't be bugged with notifications.

ggl-tommy-fritz, 2020-01-13:
We have a use case where the users device is used to open a door from a wall mounted reader via bluetooth. In this case, close proximity is required for background usage.

Perhaps close proximity is a reasonable security requirement for background usage, if that's even feasible?

nuxodin, 2020-04-21:
It would be great, you could realize a (COVID-19) "contact tracing" app as a webapp.

scheib, 2020-04-23:
Technically, it would be very interesting if we could do so given the API capabilities and other applications.  There would be significant privacy concerns to work through first.  This isn't a near term goal.

Practically, to address the pandemic, it is more appropriate to have operating systems offer this service in a power and privacy efficient way.  There is work underway to do this: https://www.blog.google/inside-google/company-announcements/apple-and-google-partner-covid-19-contact-tracing-technology/

eduarhasanaj, 2021-06-23:
Is there any progress in this feature?
We have a case where a background script of an extension requires a persistent Bluetooth connection for exchanging sensitive data locally.
I understand that there are many security concerns but cannot extension reviewers increase the review time(also costs) for such functionalities?

mtom55, 2021-07-31:
So imagine the scenario, where delivery orders are sent in and you have to print order labels to a bluetooth thermal printer.

The situations I am considering are: 
1.  If the user has switched apps, being able to receive a packet either by fetch or some server push mechanism,  then  render an image (canvas toDataURL) convert that into the printers format and then send that via bluetooth

2. being able to start that process up in the background even if the OS has decided to suspend the app

(real scenario is a little bit more complex involving multiple devices, multiple receipt printers and a label printer)

reillyeon, 2021-08-02:
Working within the current limitations I would recommend using the push notification to alert the user that they need to switch back to the application in order to print a new receipt. I recognize this isn't ideal.

sleien, 2021-11-08:
We have another use case.

We only need to know which BT devices are near. No connection is needed. Our goal is to start macros if specific IDs are close by.

mtom55, 2021-12-03:
I'd like to expand on the scenario above, as I think it's a good explanation about scenarios that Web-Bluetooth can't solve which native APIs can.  

**Scenario**
In each shop we have a POS terminal (either an ipad or an Android tablet) with a POS software running as a PWA.  

There are two (or more) printers:
1. A Label Printer
2. A Thermal Receipt Printer

**Association**
The ideal situation in the PWA is to setup and associate the PWA with each of the devices ONCE, and then have them continue to work.  It becomes less ideal the more times the user has to go through the association process.

**Printing**
When an order is placed on the system, labels are printed using the label printer so they can be stuck on the box and a receipt is printed on the thermal printer. In this case the PWA is active and can just use web-bluetooth as normal as long as both devices are connected.

**Online Order**
Online orders are received through the server, while the application is running it's possible to open a web-socket connection to receive the online order and then immediately print both a label and a receipt when one is received.  The issue is when the application is not active as the staff often need to switch to other apps. 

The ideal solution would be to use PUSH API to wake a service worker, which can immediately connect to the bluetooth printers and print.  This way if the device is idle or a user is using another application the printing process can continue uninterrupted.

Note that this entire process is possible within native apps, but is not possible on web-bluetooth.  I realize there are possible privacy/security concerns, but I think a good general stance is if you can do it on native, you should be able to do it on the web. If the privacy/security risks are too big to be mitigated by any process then that functionality should also be removed from native.

altjz, 2021-12-29:
I can't understand why `Worker` doesn't support WebBluetooth.

I saw `Worker` supports `WebSerial`.

[WorkerNavigator.serial](https://developer.mozilla.org/en-US/docs/Web/API/WorkerNavigator/serial)

I have a web app need to access my company's product.

The web app is able to access my device with `WebSerial` in worker, but it can't do the same thing with `WebBle` in worker. Wired!!

reillyeon, 2022-01-04:
@altjz, I've filed #571 to track `Worker` support separately from Service Worker support.

foxt, 2023-07-09:
It'd be nice to have a way to bypass the slowing/pausing of unfocused tab. If you have a bluetooth device that say, needs data read/written to every 100ms, there might be an issue if the user goes away to look at something else.

KOLANICH, 2023-07-10:
I'm absolutely against of Service Workers (which means malware persistence and background activation), and I'm especially against of allowing Service Workers to access Bluetooth, USB, MIDI and so on. Without proper explicit user-vetting of every action with such dangerous APIs they can hecome a privacy and security threat. User-vetting of every action by a service worker (and of every event sent to the service worker by browser!) defeats the purpose of Service Workers. And don't tell me that modern users won't install malware if they are told "install malware or go away from my website" by a powerful and essential website.

PavelCibulka, 2023-07-10:
> I'm absolutely against of Service Workers

You have to considere, that websites are replacing most of the classical EXE applications. So web browset need service workers for manage label printers, barcode scanners, scales etc.

foxt, 2023-07-10:
@KOLANICH I feel like service worker access is definately tricky. I feel like dedicated [`Worker`](https://developer.mozilla.org/en-US/docs/Web/API/Worker/Worker) access, which runs in the "background" but belongs to a specific document. 

Infact, the entire browser security model is tricky, vendors try their best to lock off functionality (mostly for security, sometimes [not](https://github.com/WICG/webhid/blob/main/blocklist.txt#L45)), but you end up in a weird situation where sometimes users can't upload files from their cloud drive, but you can download ransomware with two clicks.

PavelCibulka, 2023-07-10:
@KOLANICH
It is working well for us. HTML + JS fine for priting label on label printer or scanening barcode from goods. There is no need for efficiency.

What is important is maintenance. It is more efficient to mantain single website than update software on 1000+ devices. Also website development is much faster than EXE development.

KOLANICH, 2023-07-10:
<details>
<summary>offtop</summary>

>EXE development
>update software on 1000+ devices

Windows is a pretty uniform platform. For software distribution one can use https://learn.microsoft.com/en-us/windows/package-manager/ .
For other systems there are also package managers and software should be distributed through repos for them.

I admit that using browsers is an easy way for onboarding new users because one can put a link to the webapp into a webpage. But for anything more serious than just a demo standalone apps are more optimal.

</details>

foxt, 2023-07-10:
<details>
<summary>OT</summary>
@KOLANICH If Windows was such a unified platform with easy software distribution (would you like my laughter in .msi, NSIS, InnoSetup, WiX .exe, Squirrel, bespoke .exe installer, AppX, AppV, MSIX, ClickOnce, .cab file, portable .zip, etc) there'd be no need for an entire industry of products that exist soely to take the applications you have and run them on the infrastructure you have.

Making a web app for certain devices IMHO, the future and how everything should be. It's so much easier to just open a web page. Plus, web pages are much more isolated, (i.e. a badly made installer cant munge the registry), and there's no residual files when you uninstall an app.

Also, if it works in a web browser, it works on Windows, Mac, Android, ChromeOS, etc. You don't have to haul an old PC out of storage because the manufacturer only shipped an EXE that only works on Windows 7.
</details>

reillyeon, 2023-07-10:
Please stay on topic. To answer the original question:

> It'd be nice to have a way to bypass the slowing/pausing of unfocused tab. If you have a bluetooth device that say, needs data read/written to every 100ms, there might be an issue if the user goes away to look at something else.

Not throttling pages with active Bluetooth connections is tracked in [a Chromium issue](https://bugs.chromium.org/p/chromium/issues/detail?id=1292362) and is relatively straightforward to implement, as we recently made a [similar change for active serial connections](https://bugs.chromium.org/p/chromium/issues/detail?id=1442957).

K0IN, 2025-05-01:
hei, any updates regarding this, I want to build an webapp for a local Bluetooth chat, and I want my users to be able close the app or run in background.

what is the current state is someone active working on this proposal?

urbanonymous, 2025-05-02:
"A PWA that runs a Bluetooth-connected service in the background, like a native app, is currently not possible on any platform."
WHY?? :(