whatwg/bluetooth #428: Can bluetooth objects be used after their originating document is refreshed?

adhodi, 2019-02-18:
Once page is refreshed, the ble device still has connection to web and can push data but client has no access to connected device object. User has to again pair already paired device. It there a way to achieve this already?

reillyeon, 2019-02-18:
[This example](https://webbluetoothcg.github.io/web-bluetooth/#example-permission-api-query) shows how `navigator.permissions.query` can be used to get a `BluetoothDevice` that the user has previously granted the page permission to access. This is not currently implemented in Chrome but there is an engineer assigned to the task.

closed 2019-02-18 by reillyeon

adhodi, 2019-02-19:
@reillyeon 
yeah, i was following that only but figured out bluetooth permission is not allowed.
Is there a link to task assigned which i can follow to track rollout of this feature?

reillyeon, 2019-02-19:
https://bugs.chromium.org/p/chromium/issues/detail?id=577953

LukasBombach, 2019-06-23:
Same issue with Firefox Quantum 67.0.4