whatwg/bluetooth #429: startNotifications sometimes doesn't bind

StephaneJuban, 2019-03-23:
Hi !

I have based my code according to the example of the heart rate measurement because I want to get value from a Characteristic in Notification mode.

Everything is working fine most of the time (9 out of 10 times). But sometimes, **even if the connection is successfully made**, when I try to bind to the notification, it doesn't work.

The subscription to the Notification is made in this function :
```
_startNotifications(characteristicUuid) {
    let characteristic = this._characteristics.get(characteristicUuid);
    console.log(characteristic);
    return characteristic.startNotifications().then(() => characteristic);
}
```

When everything is OK, I can see in the console that `BluetoothRemoteGATTCharacteristic` has a `value : DataView(2) {}` Otherwise, when it's not working it has a `value : null`

Console output :
[image]

First, why do you think this happens ? Can this be because my BLE device is sending data/advertising too fast ?
I would like to be able to retry automatically, if I detect that the value is null for example.

How would you approach this ?

_I'm using Chrome but this happens on all device (Android smartphone, Mac, tablet)._
How would you approach this ?

reillyeon, 2019-03-25:
`characteristic.value` will only be populated when the `oncharacteristicchanged` event is fired and when the Promise returned by `readValue()` is resolved. In your code are you waiting for one of these events before checking the value? Just waiting for the Promise returned by `startNotifications()` to resolve is not enough to guarantee that a value is available yet.

StephaneJuban, 2019-03-25:
Thanks for your answer !
To sum up, in my code the notification connection is started with :
`device._startNotifications(characteristicUuid).then(handleHeartRateMeasurement)`

And the listener :
```
function handleHeartRateMeasurement(heartRateMeasurement) {
  heartRateMeasurement.addEventListener("characteristicvaluechanged", event => {
    var value = device.parseValue(event.target.value);
    // Use my value
  });
}
```

reillyeon, 2019-03-25:
So based on your snippets above the behavior you are observing is expected.

In `_startNotifications()` you log the value of `characteristic` but at this point you aren't guaranteed to have received a value for this characteristic yet and so `characteristic.value` may or may not be `null`. If you log `event.target` in your event listener you should always see `value` populated with a valid `DataView`.

Can you confirm that this is what you observe?

StephaneJuban, 2019-03-25:
You mean like this ?
```
function handleHeartRateMeasurement(heartRateMeasurement) {
  heartRateMeasurement.addEventListener("characteristicvaluechanged", event => {
    console.log(event.target);    
    var value = device.parseValue(event.target.value);
    // Use my value
  });
}
```

In this case yes. I have no problem to retrieve a value in the listener.

My issue is that sometimes, this listener is never called, so I assume this is because the event `characteristicvaluechanged` never happens, and I guess this happens when the binding to the notification is not properly made ?! Or because I start the listener before receiving the first `characteristic.value` ?

reillyeon, 2019-03-25:
It is safe to add the event listener before receiving the first value. You don't even need to call `startNotifications()` first. It just won't be called until the browser receives a new characteristic value.

Are you sure that the promise returned by `startNotifications()` is resolving? Adding a log statement to the beginning of `handleHeartRateMeasurement()` should answer this question.

StephaneJuban, 2019-03-26:
I added a log :
```
function handleHeartRateMeasurement(heartRateMeasurement) {
console.log("HELLO");
  heartRateMeasurement.addEventListener("characteristicvaluechanged", event => {
    console.log("HELLO bis");
    var value = device.parseValue(event.target.value);
    // Use my value
  });
}
```

"HELLO" is always displayed, but sometimes, randomly "HELLO bis" is not displayed, and so I receive no value.

---

So if you would first add the event listener before receiving the first value, would you do something like this ?
```
device.connect().then(() => device._startNotifications(characteristicUuid))

_startNotifications(characteristicUuid) {
    // Get the Characteristic
    let characteristic = this._characteristics.get(characteristicUuid);
    console.log("before event listener")

    // Start to add event listener for new value for this Characteristic
    characteristic.addEventListener("characteristicvaluechanged", event => {
      var value = device.parseValue(event.target.value);
      console.log(value)
    });

    console.log("after event listener")
    // Subscribe to the notifications of this Characteristic
    return characteristic.startNotifications().then(() => characteristic);
}
```

Like this I get the `value` in my console but sometimes, it still doesn't print anything _(except "before event listener" and "after event listener")_

But I don't get how I can receive new value if I don't use `startNotifications()` of my Characteristic ?

AndriyHromyak, 2019-04-15:
i got also problem with startNotifications. 
I got device, when clicking on device button to awake it, it will wait for pairing and will show a PIN to insert, if its new device on PC everything works fine, but if it is known device and i push button in device to pair transmitting PIN browser just ignores it and pairs without prompting popup with input for pin, and so startNotification fails with error. Any workarounds?
P.S. if i manually remove device from PC all works fine till next time device will transmit error.

reillyeon, 2019-04-15:
Since this is a Chromium issue and not a problem with the Web Bluetooth spec I'm closing this issue. Chromium [issue 920989](https://bugs.chromium.org/p/chromium/issues/detail?id=920989) may be a similar problem which we are investigating. When filing issues please include information on the platform (Windows, macOS, Chrome OS, Android) you are using.

closed 2019-04-15 by reillyeon