whatwg/bluetooth #431: [Feature Request] generate found devices in app UI, not popup

foureyedraven, 2019-04-04:
Based on what my team has said, generating the list of found BLE devices in the UI we build, without the need for the popup, is currently not possible. We would like our users to have the ability to connect to multiple BLE devices at once without having to approve each one in a pop up.

Could this be possible?

Thank you

reillyeon, 2019-04-04:
Closing this as a duplicate of #42. The list of found devices is rendered in a popup and not provided directly to the app so that it can present its own UI because the user agent needs to obtain user consent before it can give the app information about any nearby devices.

foureyedraven, 2019-04-04:
I'm not sure a feature request about allowing devs to create their own UI for users to connect to devices the same as allowing connection to multiple devices. Though, of course having multiple devices at once does follow naturally.

reillyeon, 2019-04-04:
The security and privacy model for this API requires that apps _not_ be allowed to create their own UI for users to connect to devices because that would leak private information before the user has given consent. The motivation for this request seemed to be so that multiple devices could be selected simultaneously which is why I considered it a duplicate. Are there additional motivations for this request?

foureyedraven, 2019-04-05:
I guess what I'm asking is, could there be ways for users to provide that consent through the UI instead? Sounds like a definite no, though?

foureyedraven, 2019-04-05:
It sounds like what I'm asking for (getting user consent through our UI) is a definite no due to the security model, so that answers my question. Thank you!

reillyeon, 2019-04-05:
Thank you for your feedback.

closed 2019-04-05 by reillyeon

g-ortuno, 2019-04-07:
We do have an issue to allow users to select more than one device in the chooser: #42

g-ortuno, 2019-04-07:
Oops seems like that was mentioned already!