whatwg/bluetooth #438: Is it possible to connect two web pages on two different devices using Web Bluetooth?

TomasHubelbauer, 2019-05-16:
I am asking this question here, because it is not a browser specific issue and I don't think it is a good fit for Stack Overflow, because I am asking about a use-case and whether something is or will be a part of the standard, not a specific coding question.

Is it, with the current spec, or will it be possible in the future versions of the spec, to have a web bluetooth page which is not only able to connect to a Bluetooth device, but also accept a connection from a Bluetooth device or another web page?

I am not that familiar with Bluetooth, so this may not be even technically possible, but I am wondering whether this use case is or will be supported, as it would have huge implications for mesh networking which is my area of interest.

Thank you

TomasHubelbauer, 2019-05-16:
Whoops, this is actually already discussed in #231! My bad. Looks like the piece of nomenclature I was missing was 'peripheral mode".

closed 2019-05-16 by TomasHubelbauer

jimmywarting, 2019-06-03:
came here with the same question :)

Resaki1, 2024-01-11:
Hey @jimmywarting, I know this might be a far shot, but did you get the WebRTC signaling using bluetooth to work? I'd be very interested in a solution :)

lortonx, 2024-04-02:
came here with the same question :)

riggs, 2024-04-03:
WebBluetooth likely won't support anything enabling this kind of communication for many years. WebNFC should, in theory, be able to exchange WebRTC Session Descriptions to enable this functionality. I doubt anyone has built this solution, yet.

alvinjiooo, 2024-04-05:
Allowing sites to communicate through BLE devices has interesting use cases. However, the current spec only defines APIs for central role operation. Defining peripheral behavior and scoping platform API support for peripheral mode requires substantial work and is not currently planned.