whatwg/bluetooth #439: Use AbortController instead of stop method

kenchris, 2019-05-21:
CC @reillyeon @domenic

reillyeon, 2019-05-21:
A brief scan of the specification shows three places where we could use an `AbortController` instead of a dedicated "stop" method,

* [`stopNotifications()`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothremotegattcharacteristic-stopnotifications)
* [`unwatchAdvertisements()`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothdevice-unwatchadvertisements)
* [`BluetoothLEScan.stop()`](https://webbluetoothcg.github.io/web-bluetooth/scanning.html#dom-bluetoothlescan-stop)

The latter two have not been shipped in any implementation and so are reasonable candidates for updating. We will likely have to continue supporting `stopNotifications()`.

jimmywarting, 2019-11-17:
trying to connect to one device, takes forever sometimes, would be nice with `device.gatt.connect({signal})` here also...

yorkie, 2019-11-18:
Supporting AbortController at WebBluetooth is a good idea for flexibility, but sometimes the original methods like `stop` looks like more convinent and friend-to-human.

gfwilliams, 2020-07-23:
+1 for AbortController on connect, but I feel that applying AbortController to everything is a *really bad* idea.

IMO AbortController should be for aborting tasks which you expect to complete in most cases and only want to abort in rare cases. The signal pattern allows you to specify the ability to abort but keeps the API tidy for the majority of cases, which is great.

By contrast, `startNotifications`, `watchAdvertisements`, `requestLEScan` are activities. They're not things that you ever expect to complete on their own, so you don't want to 'Abort' them. You just want an easy way to stop them.

If you use AbortController then if you accidentally call `watchAdvertisements` with no signal, you're stuck. There's *no way* to stop scanning for advertisements, even if you have a reference to the device itself.

Literally the only way to stop the advertisements coming in is to force the page to refresh - which feels to me like a big design problem - same applies if `stopNotifications` or `BluetoothLEScan.stop()` were removed

jyasskin, 2020-07-23:
I haven't thought about `AbortController` vs `stop()` enough to have strong opinions, but @gfwilliams, the worry that the program will "accidentally call `watchAdvertisements` with no signal" is no different from worrying that the program will fail to call `.stop()`. In both cases, at the point the programmer realizes that the program needs to stop scanning, they can modify the program to do what it needs to do to stop scanning. We don't modify programs while they're running or persist this part of their state to storage, so the ability to forget to pass the signal in one run doesn't make it any harder to pass the signal in the next run.

scheib, 2020-08-12:
@gfwilliams point above does apply in interactive coding.  The Espruino projects such as puck.js encourage interactive coding when developing a program to experiment with how things work and what approach to take.

jimmywarting, 2020-08-13:
I don't have strong opinion abort AbortController vs stop() either.

maybe a rule of thumb:
if something returns a promise and may take a wile, then a abortSignal could be an option
if you do however get a instance of something then you may have the opportunity to stop it

could be like comparing XMLHttpRequest and fetch
with XMLHttpRequest you get a instance back immediately and can abort something with `xhr.abort()` that's all fine but with fetch you have no other option than to use AbortSignal