whatwg/bluetooth #440: disconnect not working on windows chrome

2524user, 2019-05-24:
Not disconnected when calling diconnect without calling stopNotifications on Windows version of chrome.(e.g reload page while receiving notification)

2524user, 2019-05-26:
Is this commit relevant?
https://chromium.googlesource.com/chromium/src/+/56fb252f0f813a126596aed8be5c70998a2e05b3%5E!

2524user, 2019-05-28:
sorry, url is wrong.
https://chromium.googlesource.com/chromium/src/+/56fb252f0f813a126596aed8be5c70998a2e05b3

beaufortfrancois, 2019-05-28:
@reillyeon may be able to help

reillyeon, 2019-05-28:
Please file a report on bugs.chromium.org with example code and step-by-step instructions for reproducing this issue. If this appears to be a regression please include the last version in which it worked as expected.

Closing this issue as it is implementation-specific.

closed 2019-05-28 by reillyeon